energia/Energia #673: Direct GPIO Access on MSP432

baettigp, 2015-07-19:
Hi All,

 

I am having trouble porting code that's running perfectly on an MSP430F5529 (launchpad) to the MSP432. If I try to access the GPIO registers directly on the MSP432, the compiler gives me the following errors:

sketch_jul14b.ino: In function 'void setupsketch_jul14b()':
sketch_jul14b.ino:20:3: error: 'P2REN' was not declared in this scope
sketch_jul14b.ino:21:3: error: 'P2OUT' was not declared in this scope
sketch_jul14b.ino:22:3: error: 'P2DIR' was not declared in this scope
make: *** [sketch_jul14b.obj] Error 1
C:\ti\energia\tools\common\bin\make returned 2

According to the MSP432 technical reference manual, http://www.ti.com/lit/ug/slau356a/slau356a.pdf P 487, 500ff the registers should be called exactly the same on the 432 as they were on the 430.

Here is a minimal sketch that shows the behavior:
-----
const uint8_t pinMask = B00011111;
uint8_t input = 0;

void setup()
{
  P2REN = (P2REN & pinMask); // disable pulls
  P2OUT = (P2OUT & pinMask); // set output to low
  P2DIR = (P2DIR & pinMask) | (B11111111 & ~pinMask);  // enable output
  P2OUT = P2OUT | (~pinMask); //LEDs on
  //P2OUT=0b11100000;
  delay(2000);
}

void loop()
{

}
-----
I am not sure why it won't work on the MSP432... I compiled the above code in energia for the MSP430F5529 and the MSP430FR5739 launchpads without problem.

Then I tried to put the Port manipulation lines of the above code in an example blink program in CCS and (other than the missing binary representations which I replaced with 0x1F?) it seems to compile just fine in CCS.
-----
//...
//   Dung Dang
//   Texas Instruments Inc.
//   Nov 2013
//   Built with Code Composer Studio V6.0
//******************************************************************************
#include "msp.h"
#include <stdint.h>

const uint8_t pinMask = 0x1F;

int main(void) {
    volatile uint32_t i;

    WDTCTL = WDTPW | WDTHOLD;               // Stop WDT
    P1DIR |= BIT0;                          // P1.0 set as output

    while (1)                               // continuous loop
    {
        P1OUT ^= BIT0;                      // Blink P1.0 LED
          P2REN = (P2REN & pinMask); // disable pulls
          P2OUT = (P2OUT & pinMask); // set output to low
          P2DIR = (P2DIR & pinMask) | (0xFF & ~pinMask);  // enable output
          P2OUT = P2OUT | (~pinMask); //LEDs on
        for (i = 20000; i > 0; i--);        // Delay
    }
}
-----

 I also tried "compile driverlib from source" and that didn't work either:

C:\ti\energia\tools\common\bin\make --no-print-directory -C C:\Users\bap\AppData\Local\Temp\build2042585571797867940.tmp\ -f C:\ti\energia\hardware\common\Makefile
CCROOT C:\ti\energia\hardware\tools\lm4f
SDKROOT C:\ti\energia\hardware\msp432/CC3200SDK_1.0.0.patch
CLOSURE C:\ti\energia\hardware\common
MAINSKETCH sketch_jul14b.cpp
C:/ti/energia/tools/common/bin/make -C C:\ti\energia/hardware/msp432/cores/msp432/driverlib clean
Das System kann den angegebenen Pfad nicht finden.
Der Befehl "sh" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
process_begin: CreateProcess(NULL, uname -s, ...) failed.
process_begin: CreateProcess(NULL, dirname "", ...) failed.
process_begin: CreateProcess(NULL, rm -rf gcc-cm4f, ...) failed.
make (e=2): Das System kann die angegebene Datei nicht finden.

make[1]: *** [clean] Error 2
make: *** [driverlib] Error 2
C:\ti\energia\tools\common\bin\make returned 2

So if anyone has any hints on how to get it to work on energia, I would be most grateful.

rei-vilo, 2015-07-19:
Have you gone through the **Port with MSP430 and MSP432 MCUs** section at the [MSP432 page](http://www.ti.com/lsds/ti/microcontrollers_16-bit_32-bit/msp/low_power_performance/msp432p4x/getting_started.page?DCMP=ep-mcu-msp-432-en&HQS=msp432#first)?

>
Ready to Get Started Porting between MSP430 and MSP432 MCUs
* Step 1: Download the MSP432 MCU Platform Porting Guide
[Porting Guide (PDF, 2.82)](http://www.ti.com/lit/pdf/slaa656)
* Step 2: Download the MSPWare Software Development Package
[Download Tool](http://www.ti.com/tool/mspware)
>

For such technical question, feel free to post also at the [43oh forum](www.43oh.com) and the [E2E forum](http://e2e.ti.com).

baettigp, 2015-07-19:
Thank you very much; The second example code I gave above which compiles without problems comes from mspware http://dev.ti.com/tirex/#/Package/MSPWare?link=MSPWare%2FDevices%2FMSP430%2FMSP432P4XX%2FMSP432P401R%2FExamples%2FC%2Fmsp432p401_1%2Fmsp432p401_1.c to which I added the port manipulation lines. That compiles without problems under CCS.

However the code which compiles for FR5739 / F5529 under energia does not compile for the MSP432 under energia (all under windows if that makes a difference).

If I do the inverse and take the code for blinking P1.0 from the MSPware example and put it into energia

-----
void setup()
{
P1DIR |= BIT0;
}

void loop()
{
        P1OUT ^= BIT0;                      // Blink P1.0 LED
        for (uint16_t i = 20000; i > 0; i--);        // Delay
}
-----
this also compiles for FR5739/F5529 but gives an error for the MSP432

C:\ti\energia-0101E0016\tools\common\bin\make --no-print-directory -C C:\Users\Pio\AppData\Local\Temp\build2871327005070934537.tmp\ -f C:\ti\energia-0101E0016\hardware\common\Makefile 
CCROOT C:\ti\energia-0101E0016\hardware\tools\lm4f
SDKROOT C:\ti\energia-0101E0016\hardware\msp432/CC3200SDK_1.0.0.patch
CLOSURE C:\ti\energia-0101E0016\hardware\common
MAINSKETCH blinky.cpp
armcl main.cpp
C:\ti\energia-0101E0016\hardware\tools\lm4f/bin/arm-none-eabi-gcc -c -Os @"C:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/compiler.opt" -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostEuropeETSI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostEuropeETSI/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostUSAFCC/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostUSAFCC/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Adafruit_TMP006/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/CogLCD/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/EduBPMKII_Screen/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/LCD_SharpBoosterPack_SPI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/M2XStreamClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/MQTTClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/OPT3001/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/OneMsTaskTimer/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/PubNub/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/PubSubClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/SPI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Servo/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Temboo/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Temboo/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/WiFi/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/WiFi/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Wire/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/aJson/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/aJson/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/ -IC:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/configPkg/ -IC:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/lib/ -IC:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/ti/ -IC:\ti\energia-0101E0016/hardware/msp432/cores/msp432/ -IC:\ti\energia-0101E0016/hardware/msp432/cores/msp432/inc/ -IC:\ti\energia-0101E0016/hardware/msp432/cores/msp432/lib/ -IC:\ti\energia-0101E0016/hardware/common -DF_CPU=48000000L -D__MSP432P401R__ -DTARGET_IS_MSP432P4XX -Dxdc__nolocalstring=1 -ffunction-sections -fdata-sections -DARDUINO=101 -DBOARD_MSP_EXP432P401R -DENERGIA=14 -D__MSP432P401R__ -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant  -fno-exceptions -fno-rtti -I "C:\ti\energia-0101E0016\hardware\tools\lm4f/include" -I "C:\ti\energia-0101E0016\hardware\common" -I "C:\ti\energia-0101E0016\hardware\common" -IC:\ti\energia-0101E0016/hardware/msp432/cores/msp432/inc/CMSIS -IC:\ti\energia-0101E0016/hardware/msp432/cores/msp432/driverlib/MSP432P4xx main.cpp -o main.obj
armcl blinky.cpp
C:\ti\energia-0101E0016\hardware\tools\lm4f/bin/arm-none-eabi-gcc -c -Os @"C:\ti\energia-0101E0016/hardware/msp432/variants/MSP_EXP432P401R/compiler.opt" -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostEuropeETSI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostEuropeETSI/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostUSAFCC/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/AIR430BoostUSAFCC/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Adafruit_TMP006/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/CogLCD/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/EduBPMKII_Screen/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/LCD_SharpBoosterPack_SPI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/M2XStreamClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/MQTTClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/OPT3001/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/OneMsTaskTimer/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/PubNub/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/PubSubClient/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/SPI/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Servo/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Temboo/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/Temboo/utility/ -IC:\ti\energia-0101E0016/hardware/msp432/libraries/WiFi/ -IC:\ti\energia-0101E00 [... cut ...]
make: *** [blinky.obj] Error 1
C:\ti\energia-0101E0016\tools\common\bin\make returned 2

If I replace "BIT0" by "0x00" I still get an error concerning the registers:

blinky.ino: In function 'void setupblinky()':
blinky.ino:7:1: error: 'P1DIR' was not declared in this scope
blinky.ino: In function 'void loopblinky()':
blinky.ino:12:9: error: 'P1OUT' was not declared in this scope
make: *** [blinky.obj] Error 1
C:\ti\energia-0101E0016\tools\common\bin\make returned 2

Thank you very much for your help

robertinant, 2015-07-20:
You will have to #include <msp.h> in your Sketch.

The below compiles and should run without problem. I will look into if including msp.h in Energia.h makes sense so that it does not have to be included explicitly. Please take care in direct manipulation of pins/ports. Energia for MSP432 is implemented on top of TI-RTOS. For digital pin manipulation this is OK but be careful with things like timers, SPI, etc since those are managed by TI-RTOS.

```
const uint8_t pinMask = B00011111;
uint8_t input = 0;

void setup()
{
    P2REN = (P2REN & pinMask); // disable pulls
    P2OUT = (P2OUT & pinMask); // set output to low
    P2DIR = (P2DIR & pinMask) | (B11111111 & ~pinMask); // enable output
    P2OUT = P2OUT | (~pinMask); //LEDs on
    //P2OUT=0b11100000;
    delay(2000);
}

void loop()
{

}
```

baettigp, 2015-07-22:
Dear robertinant,
Thank you very much for your help, with this my demo-program (as well as the bigger code) compiled and ran without problems.

closed 2015-07-22 by baettigp