energia/Energia #674: CC3200: sprintf() doesn't work

rei-vilo, 2015-07-20:
`sprintf(bufferOut, "%8.1f oC %8.1f b ", t1, p);`doesn't work on CC3200:

>
     0.0 oC      0.0 b 
        23.8999996185 oC        1000.6900024414 b
>
    -0.0 oC      0.0 b 
        23.8999996185 oC        1000.7399902343 b
>

Output is fine on a MSP432, 

>
    22.6 oC  1000.8 b 
        22.6000003814 oC        1000.7500000000 b
>

Here's the basic code used

```
    myBMP180.get();

    float t1 = myBMP180.temperature();
    float p  = myBMP180.pressure();

    memcpy(bufferOut, 0x00, sizeof(bufferOut));
    sprintf(bufferOut, "%8.1f oC %8.1f b ", t1, p);
    Serial.println(bufferOut);

    Serial.print("\t");
    Serial.print(myBMP180.temperature(), DEC);
    Serial.print(" oC\t");
    Serial.print(myBMP180.pressure(), DEC);
    Serial.println(" b");

```

rei-vilo, 2015-07-20:
... but `sprintf` works with integers!

```
    uint32_t temp_int = (uint32_t)t1;
    uint32_t temp_dec = ((uint32_t)(t1 *100)) %100;

    sprintf(bufferOut, "2- %6i.%02i oC", temp_int, temp_dec);
    Serial.println(bufferOut);
```

rei-vilo, 2015-09-09:
Maybe a better solution at http://forum.43oh.com/topic/3457-printf/?p=30296

robertinant, 2015-09-23:
I just tried this and it seems to work OK for me with the following Sketch:

```
void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello, World!");
}

char buf[128];

void loop()
{
  float f = PI;
  sprintf(buf, "%8.4f", f);
  Serial.println(buf);
  delay(1000);
}
```

rei-vilo, 2015-09-23:
Yes, on Energia 16 it works, but not in CLI. Surprisingly, it also works in CLI for CC3200 EMT.

rei-vilo, 2015-09-24:
After in-depth investigation with `arm-none-eabi-objdump`, `wiring.c.o` needs to be before `print.cpp.o` for a correct `sprintf()` function.

I've modified my CLI makefile so case closed. Thank you for your help!

closed 2015-09-24 by rei-vilo

Vaitheeswaran-G, 2018-02-22:
What is sprintf, client.print, client.stop?  Help please!!!