energia/Energia #678: UART and low power

battosai30, 2015-07-21:
Hi everyone,

Playing with low power, a MSP40FR5969 launchpad and a SIM808, I found that it's impossible to go low power if you call just once Serial.begin() command. The fact is that Serial.end() function does nothing, especially restore RX and TX pins to I/O state. 

Adding this to end() function in HardwareSerial.cpp solved the problem : 

	pinMode_int(rxPin, 0);
	pinMode_int(txPin, 0);

So in my case, the consumption dropped down from 700µA to 2.7µA (both using sleep() command), doing this modification and using this code : 

 Serial1.begin(9600);
//do my stuff with my SIM808
 Serial1.end();
setInputLow(); //configures all my pins output and low
sleepSeconds(60);

It may depends on the uart hardware configuration. For example, debug UART only drop ~4µA. My SIM808 module as a level converter (MCU side is powered from a MSP430's pin).

StefanSch, 2015-07-22:
this is a good point and also a philosophic questions.
What should happen at and Serial end and where is the current coming from.
I assume that the current is coming from your external circuit which load the MSP430 TX pin (which is in UART idle - logic high state). If the Serial end would switch this pin to the input state it would float a least from the MSP430 side and can also generate a high current through the floating input state.
So in the case the current you see is generated from the external curcuit, my personal preference would be to keep the pin in the special function (UART) mode and have the output defined and let the user make a different setting (like you did), instead of defaulting it back to input with the risk of a floating input to a connected communication partner.
But that is just my opinion.

battosai30, 2015-07-22:
The problem is, from the moment you used Serial.begin() you're done ... You simply can't get back to original state with high level functions. You have to directly play with registers (PxSELy). 

For me, end() would have to restore everything like you never used begin(). 

Finally, end() doesn't really have an utility, basically it's just a flush().

spirilis, 2015-07-22:
using pinMode() on the UART pins after Serial.end() won't return it to normal? (it should)

battosai30, 2015-07-23:
I think you're right, I will try this afternoon