energia/Energia #684: CC3200 on Mac: MacBook freezes for 3 minutes when disconnecting the CC3200

rei-vilo, 2015-07-22:
MacBook freezes for 3 minutes when disconnecting the CC3200.

```/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/cc3200prog 166988 25 jui 19:49 ```
>
/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/cc3200prog /dev/cu.usbserial-cc3200B Builds/embeddedcomputing.bin
Open UART /dev/cu.usbserial-cc3200B
open UART success
Getting storage list
Bootloader Version: 3
Silicon version ES1.32 or higher
Bootloader version is 2, 0, 3, 2
It's a CC3200 device: PG1.32
BlockSize is 4096, number of blocks is 64
erasing 1 blocks starting from  4
Switch to NWP bootloader complete
Load common boot command for PG1.32
Bootloader version is 2, 0, 3, 4
BlockSize is 4096, number of blocks is 16
erasing 12 blocks starting from  0
erasing file "/sys/mcuimg.bin"
deleting file "/sys/mcuimg.bin"
erase file completed
Downloading file "/sys/mcuimg.bin" with size 45504
>

robertinant, 2015-09-22:
I have not been able to reproduce this. What version of OS X was this on?

rei-vilo, 2015-09-22:
Mac OS X 10.10.4 and 10.10.5.

Occurs even after a clean install of Mac OS X 10.10.5, Energia 16 and drivers.

robertinant, 2015-09-22:
Thanks! Will try to reproduce.

rei-vilo, 2015-09-24:
Occurs when the CC3200 is printing on the Serial port.

Looks like reminiscent of the same issue with the MSP430G2: I guess another issue with the USB drivers on Mac OS X.

Let's wait and see what Mac OS X 10.11 El Capitan brings and how it manages this specific point.

robertinant, 2015-09-24:
I have seen the issue just now. And indeed, keyboard / mouse freeze  when the CC3200 is unplugged when the Serial port is open. It seems that this is not a Serial driver (FTDI Driver issue). I just saw that when it freezes the kernel reports a USB <-> JTAG/SWD terminate timeout. JTAG uses HID and hence the timeout error seems to be the cause of the keyboard/mouse freeze. I will report to TI XDS110 maintainers for them to look into.

rei-vilo, 2016-01-31:
Seems to have been fixed with Mac OS X 10.11.3, although other issues have appeared with other boards.

closed 2016-01-31 by rei-vilo