energia/Energia #687: CC3200 WiFi: How to Get the MAC Address

rei-vilo, 2015-07-26:
Here's a simple function to get the MAC address of the CC3200. 

Would be great to integrate it to the WiFi library along with the `getRemoteIP()` as per #621.

```
    uint8_t macAddressVal[SL_MAC_ADDR_LEN];
    uint8_t macAddressLen = SL_MAC_ADDR_LEN;
    sl_NetCfgGet(SL_MAC_ADDRESS_GET,NULL,&macAddressLen,macAddressVal);
    MACAddress MAC(macAddressVal);
    Serial.print("MAC = ");
    Serial.println(MAC);
```

robertinant, 2015-08-26:
This would be an non standard WiFi library API. What is the need for getting the MAC address of the CC3200?

rei-vilo, 2015-08-26:
To ease configuration, as some routers require the list of the MAC addresses to allow access.

closed 2015-08-26 by rei-vilo

rei-vilo, 2015-09-02:
Added ** GetMacAddress ** example at https://github.com/energia/Energia/tree/master/hardware/cc3200/libraries/WiFi/examples/GetMacAddress