energia/Energia #688: CC3200 WiFi: Add Exosite Library

rei-vilo, 2015-07-26:
The [Exosite](http://ti.exosite.com) was introduced with great fanfare for the [TM4C129 Connected LaunchPad](http://www.ti.com/tool/EK-TM4C1294XL). 

Since them, Exosite has moved to version 2. and the CC3200 is here with WiFi.

Wouldn't be nice to have an Exosite library for the CC3200?

Reference for Arduino at [exosite-garage](https://github.com/exosite-garage)/[arduino_exosite_library](https://github.com/exosite-garage/arduino_exosite_library)

rei-vilo, 2015-09-19:
Easy port of the Arduino flavour to Energia.

closed 2015-09-19 by rei-vilo

FarahIssa, 2016-04-08:
Hello,

Do you have any idea if Energia Exosite library for CC3200 is available?

Thank you,

rei-vilo, 2016-04-08:
The Exosite library is pretty standard, so using it with the CC3200 shouldn't raise any issues.

FarahIssa, 2016-04-08:
I tried importing both a library for arduino and for msp430 into the libraries of CC3200. But, after I did so I couldn't even launch energia.
So, I don't know how to solve this issue.
Any help is appreciated

martincastrocm, 2016-06-25:
Hi fellows, does the mentioned library for arduino works out with the TM4C129 Connected LaunchPad through energia?

Thanks in advanced,