energia/Energia #692: MSP432: 14-bit ADC limited to 10-bit?

rei-vilo, 2015-08-04:
The MSP432 data sheet stipulated 14-bit ADC.

However, maximum value for `analogRead()` gives 1023, or 10-bit.

How to use the 14-bit resolution?

robertinant, 2015-08-04:
You can set it with analogReadResolution(). The implementation is however not entirely compatible with the Arduino API. The Arduino API allows it to be set to any int value. It is then mapped accordingly during an analogRead(). The msp432 implementation only allows for 8, 10, 12 and 14. Any other value will result in the resolution to be set to 14 bits. I'll look into what it takes to get this changed to behave the same as the Arduino API.

rei-vilo, 2015-09-07:
See application for bb1c141c08 Added analogReadResolution() for MSP432

closed 2016-08-29 by robertinant