energia/Energia #696: Sharp_BoosterPackLCD_SPI Freezes on MSP432

rei-vilo, 2015-08-10:
The example Sharp_BoosterPackLCD_SPI for the Sharp Memory Display BoosterPack freezes at the `delay()` function, either the initial `delay(1000);` in `setup()` or the `delay(200);`at the en of `loop()`.

It seems to be related to `OneMsTaskTimer`.

rei-vilo, 2015-08-10:
Let's use the RTOS `Clock` element from the Galaxia library instead!

See commit 8be91f59b5 on pull-request #697 on branch [issue_696](https://github.com/energia/Energia/tree/issue_696)

rei-vilo, 2015-08-11:
I've enriched the library with reverse display and orientation selection. 

[image]

[image]

StefanSch, 2015-08-12:
i like the new features - they are really great.
For the timer implementation i have two suggestions:
- Can we use a define which identifies that the compile is running on EMT to switch over to the Galaxia library - this would require that this lib is available for all EMT.
- I also favor the suggestion from Robert to make the decision inside the OneMsTaskTimer. This would make the usage and example easier as this would always be the same.

And finally if this changes would be in the update can go into:
.\libraries\  folder, so that it goes into all Launchpad system.

~ Stefan

rei-vilo, 2015-08-12:
@StefanSch

I agree with both suggestions. 

> * Can we use a define which identifies that the compile is running on EMT to switch over to the Galaxia library - this would require that this lib is available for all EMT.

A specific `ENERGIA_MT` flag would ease readability and maintenance. See ticket #683 **Create Compilation Variable `ENERGIA_MT`**.

> * I also favor the suggestion from Robert to make the decision inside the OneMsTaskTimer. This would make the usage and example easier as this would always be the same.

Sure! The modification with the use of `Clock` are minimal on the **Sharp_BoosterPackLCD_SPI** file, so they could easily be migrated to **OneMsTaskTimer**.

rei-vilo, 2015-08-12:
What the quick-and-dirty fix allowed me to do!

[image]

robertinant, 2015-09-22:
This has been fixed with commit: 753ea3ce2b8606c1fb796df1e687014569964519
OneMsTaskTimer now uses the SysBios Clock module.

closed 2015-09-22 by robertinant