energia/Energia #698: Wrong I2C Pins for MSP430 with MKII BoosterPack

vmanohar, 2015-08-10:
https://github.com/energia/Energia/blob/master/hardware/msp430/variants/launchpad/pins_energia.h

In pins_energia.h, lines 44 and 45 are currently:

static const uint8_t TWISDA  = 14;  /* P1.6 */
static const uint8_t TWISCL  = 15;  /* P1.7 */

Instead, they should be:

static const uint8_t TWISDA  = 10;  /* P3.0 */
static const uint8_t TWISCL  = 9;  /* P3.1 */

In addition, lines 48 and 49 are currently:

#define TWISDA_SET_MODE  (PORT_SELECTION0 | PORT_SELECTION1 /* | INPUT_PULLUP*/) /* do not enable the pull ups for this device */
#define TWISCL_SET_MODE  (PORT_SELECTION0 | PORT_SELECTION1 /* | INPUT_PULLUP*/)

They should be:

#define TWISDA_SET_MODE (PORT_SELECTION0| (PM_UCB0SDA << 8)
#define TWISCL_SET_MODE (PORT_SELECTION0| (PM_UCB0CLK << 8)

Making these changes will allow the MKII BP to be compatible with MSP430 LPs by using pins 9 and 10 for I2C instead of 14 and 15.

rei-vilo, 2015-08-10:
There have been two standards for the BoosterPacks. The most significant change has been the I²C port migrating from pins `15` + `16` to pins `9` + `10`. Each of the large family of the MSP430 LaunchPads comply with one or another of those standards. Not an easy task as this requires to be done on a LaunchPad by LaunchPad basis.

Please refer to #298 and #322.

StefanSch, 2015-10-09:
fixed with issue_708

closed 2015-10-09 by StefanSch