energia/Energia #700: CC3200 and MSP432: Pervasive Display e-Paper BoosterPack

rei-vilo, 2015-08-12:
The [Pervasive Display e-Paper BoosterPack](http://embeddedcomputing.weebly.com/pervasive-display-e-paper-epd-extension-kit.html) is based on e-ink and only requires power during the update. Perfect for ultra-low cost solutions with the MSP432!

Latest version **v.231** no longer requires a `PWM` input, contrary to previous release **v.110**.

The updated code is available at [rei-vilo / gratis / CC3200 and MSP432](https://github.com/rei-vilo/gratis/tree/CC3200-and-MSP432).

It works fine on MSP430F5529, LM4F120, TM4C123 but fails on MSP432 and CC3200.
* MSP432 (EMT): no update of the screen
* CC3200 (EMT and non-EMT): only first update works but fails after.

My update 2015-08-01 fixed pins numbers and solved conflict of variable names.
>
* Added #include Energia.h
* For Energia, changed pin names to pin numbers (pin names are deprecated)
* Changed CS for EPD_FLASH_CS to avoid conflicts with Energia
* Works on MSP430F5529, LM4F120, TM4C123
* Fails on MSP432 and CC3200
* Recognise the FPL material type by the rear labels
* Use branch [rei-vilo / gratis / CC3200 and MSP432](https://github.com/rei-vilo/gratis/tree/CC3200-and-MSP432)
>

I've also opened a ticket at https://github.com/repaper/gratis/issues/26

rei-vilo, 2015-09-20:
There seems to be a problem with SPI on the MSP432.

See [Provide Implementation for CC3200 and MSP432 LaunchPads on Energia #26](https://github.com/repaper/gratis/issues/26#issuecomment-141774107) on the [repaper/gratis](https://github.com/repaper/gratis) repository.

>
The V231_G2 now supports MSP432P401R via the MSP-EXP board.
>
There is one problem though, the repaper board cannot be directly mounted as the PANEL_ON pin which would be connected to P3_6 cannot be driven high. **In fact after a SPI.begin() call it seems that UCB2 is activated and pins P3_5, P3_5 and P3_7 are all assigned to this peripheral.**
>
As a workaround the refactored I/O library assing the PANEL_On to P5_1 on this board, so use a cable connection that connects the PANEL_ON(Red,11) to P5_1 on second row of pins; rthe remaining wires are to their original pins.
>

rei-vilo, 2015-09-20:
* Pins Names and Numbers

Name | Number | EPD
---|---|---
`P3_5` | `32` |
`P3_7` | `31` |
`P3_6` | `11` | `panel ON`

rei-vilo, 2016-01-09:
Here are the last results with film v231 and Energia 17:

* Works on MSP430F5529 and CC3200 non-EMT
* Fails on CC3200 EMT and MSP432 EMT

rei-vilo, 2016-01-10:
I managed to get it work on CC3200 EMT, after fixing the #822 **CC3200: Inconsistent Resolutions Between Non-EMT and EMT** with this modification on `S5813A.cpp`.

```
long S5813A_Class::readVoltage(void) {
	long vADC = analogRead(this->temperature_pin);

// Added for CC3200 EMT    
#if defined(ENERGIA_MT)
#if defined(__CC3200R1M1RGC__) || defined(__CC3200R1MXRGCR__)
    vADC *= 4;
#endif
#endif
// End
/*
    Serial.print("analogRead=");
    Serial.print(vADC);
    Serial.print("\treadVoltage=");
    Serial.println(REV_PD((vADC * ADC_MAXIMUM_mV) / ADC_COUNTS));
 */
	return REV_PD((vADC * ADC_MAXIMUM_mV) / ADC_COUNTS);
}
```

Still no luck on the MSP432.

>
Temperature = 19 Celsius
dc_state = 0x0
EPD: DC failed
>

rei-vilo, 2016-01-10:
SPI `mode 3` is indeed activated in MSP432. Still no display.

yeshymanoharan, 2017-06-25:
are there any solutions for this? I have tried to get the display working with the MSP432 and it fails on the EPD_display_init function with the EPD_initialize_gpio line.