energia/Energia #701: MSP432P410R on Mac OS X 10.10.4: Connection Issues

rei-vilo, 2015-08-13:
As for #693 **MSP430F5529 on Mac OS X 10.10.4: Connection Issues**, I'm experiencing issues with the MSP432 LaunchPad since I4ve updated to Mac OS X 10.10.4.

Mac OS X 10.10.4 seems to have changed something compared on USB management with Mac OS X 10.10.3.

>
/Applications/IDE/Energia.app/Contents/Resources/Java/tools/common
/DSLite/DebugServer/bin/DSLite -c /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/MSP432P401R.ccxml -f Builds/embeddedcomputing.elf
Initializing debug probe...
Configuring Debugger (may take a few minutes on first launch)...:
Initializing Register Database...
Initializing: CS_DAP_0
Executing Startup Scripts: CS_DAP_0
Initializing: CORTEX_M4_0
Executing Startup Scripts: CORTEX_M4_0
Connecting...
GEL Expression: OnTargetConnect():
finished
Info: CORTEX_M4_0: GEL Output: Memory Map Initialization Complete
Info: CORTEX_M4_0: GEL Output: Halting Watchdog Timer
Info: CORTEX_M4_0: Warning: Your MSP432P401R material is out of date. For the latest device with additional functionality, please order new samples from www.ti.com/product/msp432p401r
Flashing program...
Loading Program: Builds/embeddedcomputing.elf:
0% Preparing ... 
0% .text: 0 of 37484 at 0x0
Error: CORTEX_M4_0: Flash Programmer: Timeout error. Device init failed!
Error: CORTEX_M4_0: File Loader: Memory write failed: Unknown error
0% Finished
Error: CORTEX_M4_0: GEL: File: Builds/embeddedcomputing.elf: Load failed.
Failed: Failed to evaluate GEL_LoadProgramOnly( "Builds/embeddedcomputing.elf" ): File: Builds/embeddedcomputing.elf: Load failed.
>

As a work-around, I need to press RESET before and until Connecting... is displayed. 

As reported earlier, upload is very slow. Any way to improve and make it faster?

rei-vilo, 2015-08-14:
Same issue on Mac OS X 10.10.5.

rei-vilo, 2015-08-18:
Other traces with a different error message.

>
Initializing debug probe...
	Configuring Debugger (may take a few minutes on first launch)...:
		Initializing Register Database...
		Initializing: CS_DAP_0
		Executing Startup Scripts: CS_DAP_0
		Initializing: CORTEX_M4_0
		Executing Startup Scripts: CORTEX_M4_0
Connecting...
	Fatal: CS_DAP_0: Error connecting to the target: (Error -1170 @ 0x0) Unable to access the DAP. Reset the device, and retry the operation. If error persists, confirm configuration, power-cycle the board, and/or try more reliable JTAG settings (e.g. lower TCLK). (Emulation package 5.1.630.0) 
Failed: Failed to evaluate GEL_Connect(): Connect failed
>

rei-vilo, 2015-09-04:
I performed a full install of Mac OS X 10.10.5 and clean install of Energia 16. It solved the problem. 

The drivers provided by Energia, CCS Mac and CCS Cloud are conflicting.

As I'm using the Energia framework, I'll refrain from installing CCS Mac or CCS Cloud.

closed 2015-09-04 by rei-vilo

sKooTr, 2015-09-18:
I have installed Energia 16 on a Mac with 10.10.5 that has never had CCS installed. I did not install any drivers (as per the instructions). I am getting "Failed to evaluate GEL_LoadProgramOnly" connection failures approximately 80% of the time I attempt to program the LP MSP432 board.

I always get two USB serial ports listed in Energia16 when the board is plugged in, but unlike the instructions, they are listed as /dev/tty.usbmodem00000001 and /dev/tty.usbmodem00000004. There is no port listed as /dev/cu.uart-XXXX or /dev/tty.uart-XXX. Clearly something has changed since the instructions were written. 

Can someone investigate the connection problem and update the instructions please?

Thanks

measley, 2015-09-24:
Looking into this issue as it has come up with a few of the Mac users I've met. Some kind of DSLite conflict with the firmware shipped with the LaunchPad and the DSLite that ships with Energia. Hopefully the team will post updated workaround instructions soon, but I imagine this will be fixed in future releases.

rei-vilo, 2015-09-24:
@measley Thank you so much for confirming my issue. 

I had no alternative but to do a clean install of Mac OS X + Energia + drivers.

In the meantime, CCS for Mac and CCS Cloud are in quarantine.