energia/Energia #702: CC3200 on Mac OS X: Connection Fails

rei-vilo, 2015-08-13:
After #693 **MSP430F5529 on Mac OS X 10.10.4: Connection Issues** and #701 **MSP432P410R on Mac OS X 10.10.4: Connection Issues**, now the CC3200 is experiencing connection issues as well.

From Energia, uploading a 60672-byte project fails, while the basic blink succeeds. I've reinstalled the `EnergiaFTDIDrivers2.2.18.pkg` included in the Energia 16 package, to no avail.

Configuration: Mac OS 10.10.4, 

>
Binary sketch size: 60672 bytes (of a 262144 byte maximum)
\>>>> Done <<<<
/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/arm-none-eabi-objcopy -O binary QR_WiFi_temperature.elf QR_WiFi_temperature.cpp.bin
cp /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/cc3200emt/configPkg/package/cfg/energia_*.rov.xs .
/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/cc3200prog /dev/tty.usbserial-cc3200B /var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build7335152084074265644.tmp/QR_WiFi_temperature.cpp.bin 
Open UART /dev/tty.usbserial-cc3200B
open UART success
Getting storage list
Bootloader Version: 3
Silicon version ES1.32 or higher
Bootloader version is 2, 0, 3, 2
It's a CC3200 device: PG1.32
BlockSize is 4096, number of blocks is 64
erasing 1 blocks starting from 4
Switch to NWP bootloader complete
Load common boot command for PG1.32
Bootloader version is 2, 0, 3, 4
BlockSize is 4096, number of blocks is 16
erasing 12 blocks starting from 0
erasing file "/sys/mcuimg.bin"
deleting file "/sys/mcuimg.bin"
erase file completed
Downloading file "/sys/mcuimg.bin" with size 71692
Open operation failed
>

rei-vilo, 2015-08-14:
Same issue on Mac OS X 10.10.5.

rei-vilo, 2015-08-14:
Seems to be size-related

>
Downloading file "/sys/mcuimg.bin" with size 62200
Open operation failed
>

but

>
Downloading file "/sys/mcuimg.bin" with size 55288
..............Download complete
>

robertinant, 2015-08-14:
"Open operation failed" means that there is not enough space in flash to accommodate the image that you are trying to upload. To overcome this issue you have to use UniFlash and format the flash. 

Robert

rei-vilo, 2015-08-14:
Thanks for the tip! But [UniFlash](http://processors.wiki.ti.com/index.php/Category:CCS_UniFlash) is only available for Windows and Linux :(

robertinant, 2015-08-14:
Unfortunately it is :-(

rei-vilo, 2015-08-15:
This point would be a nice addition to the wiki page at http://energia.nu/cc3200guide/.

rei-vilo, 2015-08-25:
Related to #456?

rei-vilo, 2015-09-04:
I performed a full install of Mac OS X 10.10.5 and clean install of Energia 16. It solved the problem. The drivers provided by Energia, CCS Mac and CCS Cloud are conflicting. Issue has been escalated to the E2E forum at https://e2e.ti.com/support/microcontrollers/msp430/msp430mcupowerplayers/f/912/p/445414/1620248#1620248

closed 2015-09-04 by rei-vilo