energia/Energia #703: Truncated output in console when compiling

rei-vilo, 2015-08-13:
The console seems to be limited in size. 

Whatever the verbose option turned on or off, first lines are removed.

For example, compiling against the CC3200 EMT floods the console with warnings.

Is there a way to obtain the full output without being truncated?

robertinant, 2015-09-22:
I am not able to reproduce this in Energia 16. All lines are shown in the console. Can you post an example output here?

rei-vilo, 2015-09-22:
When the output for compilation and upload is larger than a given size, first lines are missing.

rei-vilo, 2015-09-23:
This also occurs with the Arduino IDE. When **Show verbose output during** **compilation** and **upload** are both selected, the trace for upload may erase the trace from compilation.

So I'm flagging this ticket as **Not Energia Specific** and closing it.

closed 2015-09-23 by rei-vilo