energia/Energia #704: MQTTClient HelloWifi  does not compile

mkarliner, 2015-08-17:
OOB HelloWifi gives the following compile error.

In file included from HelloWiFi.cpp:28:0:
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/MQTTClient/MQTTClient.h: In member function 'int MQTT::Client<Network, Timer, MAX_MQTT_PACKET_SIZE, MAX_MESSAGE_HANDLERS>::cycle(Timer&)':
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/MQTTClient/MQTTClient.h:535:21: error: 'stdout' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/MQTTClient/MQTTClient.h:535:21: error: there are no arguments to 'fprintf' that depend on a template parameter, so a declaration of 'fprintf' must be available [-fpermissive]
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/MQTTClient/MQTTClient.h:535:21: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)

rei-vilo, 2015-08-17:
Is this error specific to a MCU, or valid for different MCUs?

In both cases, which ones?

mkarliner, 2015-08-17:
MSP430 and yes, I'm using the cc3100.

Mike

rei-vilo, 2015-08-17:
Which version of the MSP430? G2553, F5529, FR4133, FR6989, ...?

mkarliner, 2015-08-17:
MSP430f5529 (25Mhz).

robertinant, 2015-09-22:
Fixed with f70ed2da495fe9603a100a66f7c7155df06bc52b. 
Replace MQTTClient.h in hardware/msp430/libraries/MQTTClient with the one here: https://raw.githubusercontent.com/energia/Energia/master/libraries/MQTTClient/MQTTClient.h

closed 2015-09-22 by robertinant