energia/Energia #707: A simple "Hello World"  does not run on TIVA-C

chalo1311, 2015-08-22:
My initial program:
_____________________________________
#define RLED RED_LED
#define GLED GREEN_LED

void setup()
{
  pinMode(RLED, OUTPUT);
  pinMode(GLED, OUTPUT);

  digitalWrite(RLED, HIGH);
  delay(1000);  
  digitalWrite(RLED, LOW);   

  Serial.begin(9600);
  Serial.print("Hello World");
 
  digitalWrite(GLED, HIGH);   
  delay(1000);  
  digitalWrite(GLED, LOW);   

}
void loop()
{
}
__________________________________

I am using a launchpad w tmc123(80 Mz).
The language is Energia.

The program ran well  only once.
The second time turns the led on and off, however 
the serial monitor does not displays the
"Hello World"
I did suspect of the board so I did ran the program
in a launchpad w lm4f120 (80Mz) and it makes the same:
turns on and off the led with the right colors and 
does not display the  
"Hello World"

I wolud appreciate someone to help me

rei-vilo, 2015-08-22:
Feel free to post your question at the dedicated forum [43oh / Energia - TivaC/CC3XXX](http://forum.43oh.com/forum/125-energia-tivaccc3xxx/).

robertinant, 2015-08-26:
I tried to reproduce but it always works for me. Make sure you have the right serial port selected in Energia. As @rei-vilo suggested. You will get quicker response to your question on the 43oh forum.

Thanks,

Robert

closed 2015-08-26 by robertinant