energia/Energia #712: SHARP96 LCD_SharpBoosterPack_SPI on cc3200 launchpad does not compile

markv1, 2015-09-03:
sharp 96 LDC booster pack lib/example does not compile for the CC3200

-fdata-sections, -mthumb, -mcpu=cortex-m4, -DF_CPU=80000000L, -MMD, -DARDUINO=101, -DENERGIA=16, -I/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200/cores/cc3200, -I/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200/variants/launchpad, -I/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200/libraries/SPI, -I/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200/libraries/OneMsTaskTimer, -I/Users/markvanderwiele/Documents/Energia/libraries/LCD_SharpBoosterPack_SPI, /var/folders/hw/sgdjxchd4fl1kpp64qwtbzz80000gn/T/build1879015637336990543.tmp/LCD_SharpBoosterPack_SPI_main.cpp, -o, /var/folders/hw/sgdjxchd4fl1kpp64qwtbzz80000gn/T/build1879015637336990543.tmp/LCD_SharpBoosterPack_SPI_main.cpp.o]
LCD_SharpBoosterPack_SPI_main.cpp: In function 'void loop()':
LCD_SharpBoosterPack_SPI_main.cpp:46:35: error: 'LCDWrapNone' was not declared in this scope
 // Add loop code
                                   ^
LCD_SharpBoosterPack_SPI_main.cpp:67:14: error: 'class LCD_SharpBoosterPack_SPI' has no member named 'setCharXY'
     for (uint8_t i=0; i<=20; i++) {
              ^
LCD_SharpBoosterPack_SPI_main.cpp:68:14: error: 'class LCD_SharpBoosterPack_SPI' has no member named 'print'
       myScreen.setXY(70,30+i,1);
              ^
LCD_SharpBoosterPack_SPI_main.cpp:70:14: error: 'class LCD_SharpBoosterPack_SPI' has no member named 'setCharXY'
     
              ^
LCD_SharpBoosterPack_SPI_main.cpp:71:14: error: 'class LCD_SharpBoosterPack_SPI' has no member named 'print'
     myScreen.setFont(1);

rei-vilo, 2015-09-03:
Which CC3200: Energia standard or Energia MT?

A development is under way for EMT. See #696.

markv1, 2015-09-03:
I am using the energy mt  0101E0016 7/8/15 from http://energia.nu/download/  on a CC3200 Launchpad  sorry I should have mentioned that

rei-vilo, 2015-09-03:
So the solution is coming... See #696.

robertinant, 2015-09-04:
This has been fixed in the master branch. You will need the files from:
https://github.com/energia/Energia/tree/master/libraries/LCD_SharpBoosterPack_SPI
https://github.com/energia/Energia/tree/master/libraries/OneMsTaskTimer

Robert

closed 2015-09-04 by robertinant