energia/Energia #713: Add addStringToObject with String to aJSON Library

rei-vilo, 2015-09-08:
Add addStringToObject with `String` to aJSON library. Up to now, it only accepts `const char* s`.

```
	void addBooleanToObject(aJsonObject* object, const char* name, bool b);
	void addTrueToObject(aJsonObject* object, const char* name);
	void addFalseToObject(aJsonObject* object, const char* name);
	void addNumberToObject(aJsonObject* object, const char* name, int n);
        void addNumberToObject(aJsonObject* object, const char* name, double n);
	void addStringToObject(aJsonObject* object, const char* name, const char* s);
```

robertinant, 2015-09-25:
This is an external library to Energia maintained here: https://github.com/interactive-matter/aJson. Could you please ask the maintainer to add this feature to the library?

closed 2015-09-25 by robertinant

rei-vilo, 2015-09-25:
Done, see https://github.com/interactive-matter/aJson/issues/74