energia/Energia #721: Extending the baud rates available in the Serial Monitor window

dpharris, 2015-09-19:
Hi --

Can we extend the baud rates in Energia/app/src/processing/app/SerialMonitor.java?
I can run the serial at higher than 115200 on Stellaris / Tiva, but can't test at that those speeds with the Serial Monitor window :-p

Is it as simple as changing: 
 "    String[] serialRateStrings = {
      "300","1200","2400","4800","9600","14400",
      "19200","28800","38400","57600","115200"
    };"

to:
    String[] serialRateStrings = {
      "300","1200","2400","4800","9600","14400",
      "19200","28800","38400","57600","115200",
      "230400", "250000", "460800", "500000",
      "921600", "1000000"
    };

I guess I can clone and build .... I will give that a try ...

Thanks
David

rei-vilo, 2015-09-20:
As a solution, use a terminal application, like 
- Putty http://www.putty, 
- CoolTerm http://freeware.the-meiers.org
- ZTerm http://www.dalverson.com/zterm/

closed 2015-09-20 by rei-vilo

dpharris, 2015-09-20:
Thanks Rei.  Although those are solutions, they are not as convenient as
the builtin Serial Monitor.  Is there a reason why adding speeds is not
desired?

Thanks again,
David

rei-vilo, 2015-09-20:
Energia is based on Arduino. Superseding Energia is always an issue, especially for maintainability. This change should come from Arduino.

dpharris, 2015-09-20:
Fair enough!  Thanks.
David