energia/Energia #722: MSP432: GPIO Issue When Using SPI

rei-vilo, 2015-09-20:
There seems to be a problem with GPIO and SPI on the MSP432.

See [Provide Implementation for CC3200 and MSP432 LaunchPads on Energia #26](https://github.com/repaper/gratis/issues/26#issuecomment-141774107) on the [repaper/gratis](https://github.com/repaper/gratis) repository.

>
The V231_G2 now supports MSP432P401R via the MSP-EXP board.
>
There is one problem though, the repaper board cannot be directly mounted as the PANEL_ON pin which would be connected to P3_6 cannot be driven high. **In fact after a SPI.begin() call it seems that UCB2 is activated and pins P3_5, P3_5 and P3_7 are all assigned to this peripheral.**
>
As a workaround the refactored I/O library assing the PANEL_On to P5_1 on this board, so use a cable connection that connects the PANEL_ON(Red,11) to P5_1 on second row of pins; rthe remaining wires are to their original pins.
>

* Pins Names and Numbers

Name | Number | EPD
---|---|---
`P3_5` | `32` |
`P3_7` | `31` |
`P3_6` | `11` | `panel ON`
`P5_1` | `33` |

robertinant, 2015-09-22:
Reproduced with the Sketch below:

```
// most launchpads have a red LED
#define LED 11
const int slaveSelectPin = 2;
//see pins_energia.h for more LED definitions
//#define LED GREEN_LED
  
// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(LED, OUTPUT);   
  SPI.begin();

}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(10);               // wait for a second
  digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
  delay(10);               // wait for a second
}
```

rei-vilo, 2015-09-22:
Referenced in #700

closed 2016-08-29 by robertinant