energia/Energia #724: analogReadResolution() not available on all LaunchPads

adrianF-TI, 2015-09-23:
This API should be enabled on all LaunchPads. Since there are different ADC resolutions available across the LaunchPad ecosystem, using the analogReadResolution() within example sketches will help improve code portability.

I tried to use this API on the MSP-EXP430F5529LP LaunchPad, but was not able to. 

Thanks!

StefanSch, 2015-09-23:
I just had a look into this but could not properly follow your arguments. Maybe you can give more details
Which Launchpad does support this function?
I tried to grep for analogReadResolution but could not find it in any Launchpad version.

spirilis, 2015-09-23:
IIRC it's a standard Arduino API feature these days.

spirilis, 2015-09-23:
https://www.arduino.cc/en/Reference/AnalogReadResolution - Added to support the ARM-based Arduino's which have a 12-bit ADC.  Good thing to implement for all LaunchPad platforms.

robertinant, 2015-09-23:
This is pretty easy to add. Question is, what should we set as the default resolution? If a user currently uses the analogRead() on say MSP430F5529 he/she gets a 12 bit value. I think best would be to leave the default resolution is is currently implemented. A user can then change the resolution with analogReadResolution();

spirilis, 2015-09-23:
It does look like the universal Arduino default is 10-bits for all platforms (even those sporting 12-bit ADCs), then you use analogReadResolution to enable something greater than 10-bits.

robertinant, 2015-09-23:
Yes, but a lot of users will be compiling their Sketches expecting the previous behavior. I do not want to break their Sketches. We could document it but the majority won't read it.

Also, the documentation (https://www.arduino.cc/en/Reference/AnalogReadResolution) does not match the implementation. It states that if you set the resolution higher than the actual resolution supported by the board it will return the value with the highest resolution supported by the board and pad the rest with 0.
However, the implementation looks like the below. Am I misunderstanding the documentation?

```
static inline uint32_t mapResolution(uint32_t value, uint32_t from, uint32_t to) {
	if (from == to)
		return value;
	if (from > to)
		return value >> (from-to);
	else
		return value << (to-from);
}
.....
uint32_t analogRead(uint32_t ulPin)
{
.....
        ulValue = adc_get_latest_value(ADC);
        ulValue = mapResolution(ulValue, 10, _readResolution);
.....
```