energia/Energia #729: Change sslConnect for connectSSL for compatibility with new WiFi library

rei-vilo, 2015-10-02:
Arduino has deprecated the WiFi library and promotes the new [WiFi101 library](https://www.arduino.cc/en/Reference/WiFi101) with the new `connectSSL()` [function](https://www.arduino.cc/en/Reference/WiFi101ClientConnectSSL). 

The Energia WiFi library provides the `sslConnect()` function, but documentation is mum about it.

* Change `sslConnect()` for `connectSSL()` for compatibility with new WiFi101 library
* Include documentation about it.

robertinant, 2015-10-02:
Thanks for bringing this up @rei-vilo!
I'll add a stub for connectSSL() to sslConnect() so that we remain backward compatible with our current implementation and will be compatible with the new WiFi101 library.

rei-vilo, 2015-10-02:
You're welcome!