energia/Energia #736: Problem with xds110 with firmware 2.2.4.2

jmgc, 2015-10-06:
In the last version of ccscloud, the xds110 from the mps432 is upgraded to the 2.2.4.2 version. When this is done, if you try to connect through Energia, the result is:

Fatal: CS_DAP_0: Error connecting to the target: (Error -263 @ 0x0) Incompatible XDS110 firmware detected.  The firmware version of the connected XDS110 debug probe does not  match the expected version. Please update the firmware using the xdsdfu  utility found in  the .../ccs_base/common/uscif/xds110 directory of  your installation. View the ReadMe.txt file there for instructions. (Emulation package 5.1.630.0) 
Failed: Failed to evaluate GEL_Connect(): Connect failed

Is there any alternative apart from downgrading the firmware?

Thanks in advance

rei-vilo, 2015-10-06:
This confirms points #701 #702.

jmgc, 2015-10-06:
I have checked with Code Composer on Linux and it also upgrades the MSP432.

rei-vilo, 2015-10-06:
I faced the same problem with CCS, CCS Cloud and Energia on Mac OS X, and with CCS on Mac OS X and Windows.

The problem is actually twofold.

- CCS, CCS Cloud and Energia don't share the same release of the driver. 
- For a given product (e.g. CCS), the different platforms (e.g. Mac OS X and Windows) don't use the same release of the driver.

jmgc, 2015-10-06:
Red Hat 6

jmgc, 2015-10-07:
I have made a test in Mac OS to see if I could upgrade easily Energia. I have made a copy of the DSLite coming from the ccscloud to the Energia directory. The DSLite has been used by energia, but it seems it uses a different set of parameters.

The Energia one uses:

```
No configuration supplied

Supported options:
  --help                      print help message
  -c [ --config ] arg         specify configuration (ccxml) file
  -f [ --file ] arg           specify program or binary file to flash
  -l [ --loadAddress ] arg    specify load address for a binary file
  -i [ --interactive ]        start interactive mode where GEL commands are 
                              read from stdin
  -s [ --loadSymbols ]        load symbols for subsequent access
  -v [ --verbosity ] arg (=2) verbosity of output [0-2]
  --pretty arg (=0)           pretty print json strings
```

while the one coming from ccscloud uses:

```
DSLite [operation] [Args...]

	load [ccxmlFile] [fileToLoad]
		Configures for the target defined in the ccxml file,
		loads the specified file and then exits leaving the 
		target running

	runWithCIO [ccxmlFile] [fileToLoad]
		Configures for the target defined in the ccxml file,
		loads the specified file and runs the program until 
		end of the application, printing stdout to the console

	help
		Prints this message
```

I assume you know it, but in Ti there is a page: http://processors.wiki.ti.com/index.php/XDS_Emulation_Software_Package that provides the latests versions for windows and linux, it is possible that it solves the problem between Linux and Windows ccs. With Mac it seems a little bit more complex.

closed 2016-08-29 by robertinant