energia/Energia #740: Provide Examples for CC2650 SensorTag

rei-vilo, 2015-10-09:
Among other

* Read data form sensors
* Display on Sharp Memory Display
* Send and receive through BLE

rei-vilo, 2016-03-13:
I've managed to read data from the HDC1000, TMP007 and OPT3001 sensors.

Last one, the BMP280, raises some issues.

* It is powered by pin 22 or `MPU PWR` or `DIO_12 `. Calling `Board_MPU_POWER` as defined in Energia.app/Contents/Resources/Java/hardware/emt/ti/runtime/wiring/cc26xx/variants/CC2650STK_BLE/Board.h results in an error.

```
pinMode(Board_MPU_POWER, OUTPUT);
digitalWrite(Board_MPU_POWER, Board_MPU_POWER_ON);
```

* It is connected through a specific I²C port on pins `DIO_09 and `DIO_08`. Calling `Wire1` raises an error.

```
TwoWire Wire1(1);
Wire1.begin();
```

rei-vilo, 2016-03-13:
To be more specific, using one single I²C port works.

Either 

```
Wire.begin()
```

or

```
TwoWire Wire1(1);
Wire1.begin();
```

works separately.

But trying and using both fails.

```
Wire.begin()
TwoWire Wire1(1);
Wire1.begin();
```

rei-vilo, 2016-03-13:
See #863.

rei-vilo, 2016-03-14:
The MPU9250 includes a DMP. The DMP is a processor that fusions the data from the accelerometer, gyroscope and magnetometer and calculates the Euler angles.

Such a library is available for Arduino at https://github.com/rpicopter/ArduinoMotionSensorExample

jkredz, 2016-03-21:
Hi rei-vilo,

I am also interested in programming the cc2650 sensor tag with Energia.  Have there been any updates or example code uploaded recently?  Could you direct me to some resources you have used that could get me started with BLE and reading data off the the sensor tag?

rei-vilo, 2016-03-21:
The libraries for BLE aren't available due to licence restrictions.

jkredz, 2016-03-21:
So are there currently no work arounds to read or respond to BLE events using the Energia platform?  I was wondering if I could make simple modifications to current sensortag program operation without programming with CCS

spirilis, 2016-03-21:
Nope.  BLE support is dead in the water, or dead on arrival should be the preferred phrase, due to TI's own legal issues.  Embarrassing, really, but using CCS is the only way to do this (and you have to grok the BLE-Stack and TI-RTOS stuff at least on a superficial level, from what I gathered).

rei-vilo, 2016-03-21:
For the moment, no.

You can download the BLE stack from TI, then try and wrap it into a C++ interface.

jkredz, 2016-03-21:
Cool, thanks for the quick responses.  Guess there is no way around learning RTOs and the BLE-Stack.  Its been a pretty steep learning curve so far for me with just a background with Arduino programming.  Aside from the software manual, RTOs tutorial videos and basic sensortag tutorial, do you have any suggested readings or sites you found helpful to help break down learning how to program the sensortag?

rei-vilo, 2016-03-21:
For RTOS,

* [LaunchPad MSP432: RTOS for Everyone](http://embeddedcomputing.weebly.com/launchpad-msp432-rtos-for-everyone.html)
* [Exploring RTOS with Galaxia](http://embeddedcomputing.weebly.com/exploring-rtos-with-galaxia.html)

jkredz, 2016-03-21:
Thanks!  Looks like some good starting points off your site.

cbmoller, 2016-04-14:
Hi @rei-vilo 
I have started exploring the sensor tag and are trying to run this example to test the MPU's capabilities. https://github.com/energia/emt/blob/a350116673c64a5ba71a034b3baaa384b888776e/src/ti/runtime/wiring/cc26xx/tests/alan/mpu9250.cpp 
when I run the code it hangs at Wire1.endTransmission(); 

Do you have an example I could follow or a possible fix? my goal is to print the readings of the accelerometer over the serial.
My code is with small modification is here:

`/*
 *  ======== mpu9250.cpp ========
 *  This code initially came from http://www.lucidarme.me/?p=5057
 */

#include <Energia.h> // Energia Wiring API
#include <Wire.h>
#include <ti/sysbios/family/arm/m3/Hwi.h>
 
//#include "app.h"

#include "MPU6500.h"

#define    MPU9250_ADDRESS            0x68
#define    MAG_ADDRESS                0x0C
 
#define    GYRO_FULL_SCALE_250_DPS    0x00  
#define    GYRO_FULL_SCALE_500_DPS    0x08
#define    GYRO_FULL_SCALE_1000_DPS   0x10
#define    GYRO_FULL_SCALE_2000_DPS   0x18
 
#define    ACC_FULL_SCALE_2_G        0x00  
#define    ACC_FULL_SCALE_4_G        0x08
#define    ACC_FULL_SCALE_8_G        0x10
#define    ACC_FULL_SCALE_16_G       0x18
 
#define LED RED_LED
#define BUZZER 23

TwoWire Wire1(1);

/*
 *  ======== I2Cread ========
 *  This function read Nbytes bytes from I2C device at address Address.
 *  Put read bytes starting at register Register in the Data array. 
 */
void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data)
{
    Wire1.begin();
    // Set register address
    Wire1.beginTransmission(Address);
    Wire1.write(Register);
    Serial.println("read data ok");
    Wire.write(1);
    Wire1.endTransmission();
    Serial.println("read end ok");
    //Wire1.end();
 
    // Read Nbytes
    Wire1.requestFrom(Address, Nbytes); 
    Serial.println("read request ok");
    uint8_t index = 0;
    while (Wire1.available()) {
        Data[index++] = Wire1.read();
    }
}
 
/*
 *  ======== I2CwriteByte ========
 *  Write a byte (Data) in device (Address) at register (Register)
 */
void I2CwriteByte(uint8_t Address, uint8_t Register, uint8_t Data)
{
[... 143 lines cut ...]
 
    // Magnetometer
    mx = Mag[1] << 8 | Mag[0];
    my = Mag[3] << 8 | Mag[2];
    mz = Mag[5] << 8 | Mag[4];
 
    // Magnetometer
    Serial.print(mx, DEC); 
    Serial.print("\t");
    Serial.print(my, DEC);
    Serial.print("\t");
    Serial.print(mz, DEC);  
    Serial.print("\t");
    // End of line
    Serial.println("");
#endif

    newTone = ax/2 + 2000;
    if (abs(newTone - oldTone) > 20) {
        tone(BUZZER, newTone, 0);
        oldTone = newTone;
    }
    delay(100);    /* otherwise get data 100 times per sec */
    
}`

rei-vilo, 2016-04-14:
Wire and Wire1 are exclusive. See #863.

cbmoller, 2016-04-15:
Hi @rei-vilo  thank you for answering.

Yes I am aware of the Wire and Wire1 conflict, i am only using Wire1 for now.
I am not doing anything to power the MPU9250, is there a pin I should set high?

EnkhbatB, 2016-07-28:
Hello,
I am working on a Sensortag board. I am using Energia last version. it is no Radio support yet for the CC2650. Could you help me? How can I porting a Bluetooth open-source on the board?

rei-vilo, 2016-07-28:
For legal reason, the BLE stack isn't available for Energia. 

However, you can register and download it from [Bluetooth low energy software stack](http://www.ti.com/tool/ble-stack).

PAk-CatchFire, 2016-09-20:
I have tried with TIVA and MPU9250.
https://github.com/PAk-CatchFire/MPU9250-Tivaware

rei-vilo, 2016-09-20:
Energia 1.6.10E18 no longer supports the CC2650 SensorTag. 

>
CC2650 has been removed from Energia due to the lack of BLE support that could not be enabled because of licensing issues with the BLE stack.
>

closed 2016-09-20 by rei-vilo