energia/Energia #746: Problem with RedBearLabs WiFimini using all PWM channels

Spacely70, 2015-10-13:
Hi,

I'm using Energia 0101E0016 and i've written this code to test the simultaneous use of the four PWM channels on a RedbearLabs WiFimini:
```
void setup()
{
  // put your setup code here, to run once:
  pinMode(6, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(15, OUTPUT);
}
void loop()
{
  // put your main code here, to run repeatedly:
  int i=0;
  for (i=0; i <= 255; i++){
    analogWrite(6,i);
    analogWrite(5,i);
    analogWrite(14,i);
    analogWrite(15,i);
    delay(10);
  }
}
```
On my board the 3rd and 4th PWM seems to have a glitch...
When some value below approx. 250 is written to PWM on pin 15 the PWM signal on pin 14 is gone
And when the value is above this value, the PWM on pin 14 is simply switched to high.
 
Can this be peproduced by anybody? Is this a bug in settings for the wifimini?
Thanks for helping!

rei-vilo, 2015-10-14:
Which environment are you using: standard Energia or MT?

Spacely70, 2015-10-14:
it's standard Energia.

robertinant, 2015-10-19:
This smells like a bug. I will have to look into this later this week.

Spacely70, 2015-11-08:
Any news on that issue?

rei-vilo, 2015-11-08:
Some PWM outputs share the same timer, hence not all PWM outputs can run together. 

See [analogWrite](/http://energia.nu/reference/analogwrite) for the mutually excluding pins.

Have you contacted RedBearLab? They are the maker of the board.

Spacely70, 2015-11-23:
Hi again,

I found that on TI's web: http://processors.wiki.ti.com/index.php/CC32xx_PWM

They write:

The general purpose timers (GPTs) support a 16-bit pulse-width modulation (PWM) mode with software-programmable output inversion of the PWM signal. As there are 4 GPTs, there are 8 possible PWM output options. Out of these, one is not available (the Timer A output of TIMERA2).

so use of four PWM's should be possible....or is it a limitation by the board not the chip?

Thomas

rei-vilo, 2015-11-23:
Please refer to the [analogWrite() reference page](http://energia.nu/reference/analogwrite/).

robertinant, 2015-11-23:
CC3200-LAUNCHXL:
Pins: 9 and 39 (both routed to GPIO_10 of the chip)
Pins: 10 and 40 (both routed to GPIO_11 of the chip)
Pins: 13 and 36 (both routed to GPIO_25 of the chip)
Pins: 19 and 37 (both routed to GPIO_09 of the chip)

You can use 4 PMW's but you have to make sure that you are not using the conflicting pins. So if you analogWrite() to 9 you should not analogWrite() to 39. Bot will have the same PWM signal.

closed 2015-11-23 by robertinant