energia/Energia #747: rf2500 issue for Mac OS X EI Capitan (have to unplug and plug again for work)

jiaguo0522, 2015-10-15:
Hi there, I just updated my Mac to OS X EI Capitan. I encountered the rf2500 issue. Everytime after I write program to the launchpad, the system cannot recognize it again unless I unplug and plug it again. Any solution? Thanks.

This is what I got from the terminal:
******************************************************************************************************
mspdebug rf2500 'prog Swap.elf'
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open interface 1 on 005
Initializing FET...
rf2500: can't receive data: Operation timed out
fet: open failed
Trying again...
Initializing FET...
FET protocol version is 30394216
rf2500: can't receive data: Operation timed out
fet: init failed
make: *** [flash] Error 255
******************************************************************************************************

rei-vilo, 2015-10-15:
Seems to be related to Mac OS X 10.11 and / or mspdebug. USB management is especially volatile on Mac OS X.

Have you contacted the developer of mspdebug at http://dlbeer.co.nz/mspdebug?

jiaguo0522, 2015-10-15:
Just sent email to him. Thanks for your reply.

dazhbog, 2016-05-26:
I am having the same problem with El Capitan and my 5529 msp430. 

I just installed the latest version of mspdebug 0.24 on top of Energia but still I have to plug and unplug my Launchpad 5529. 

It seems to be a CDC/VCP driver issue rather an mspdebug?

mcafzap, 2016-05-26:
Almost certainly an OSx issue; I hit this issue some time ago and have been putting-up with it for ages. I also found it was intermittent and would occasionally work for 2 or 3 times without unplugging.

rei-vilo, 2016-05-26:
The USB drivers are unstable with Mac OS X 10.11. Each release changes the way they are handled.