energia/Energia #749: CC3200 LaunchPad WiFi.begin() hanging on when internally powered

tom-giesberg, 2015-10-22:
When I use J20 (Battery Power) or P1.1 (Booster Pack Power Supply)) to power my CC3200 LaunchPad, WiFi.begin() causes setup() to go into an infinite loop. If I run power in through the micro USB connection all is well.

May be related: https://github.com/energia/Energia/issues/655

Test code:
```c
#include <WiFi.h>

const int ledPin =  RED_LED;

void setup() {
  pinMode(ledPin, OUTPUT);      
  blink(ledPin, 1000);
  WiFi.begin("your_ssid", "your_pw");
}

void loop() {  
  blink(ledPin, 100);
}

void blink(int pin, int time) {
  digitalWrite(pin, HIGH);
  delay(time);  
  digitalWrite(pin, LOW);  
  delay(time);
}
```
Pass - one slow blink, followed by fast blinking
Fail - slow blinking

rei-vilo, 2015-10-22:
Which framework: standard Energia or Energia MT?

Does the same sketch run correctly when the LaunchPad is powered by USB?

Does the battery deliver enough power?

tom-giesberg, 2015-10-22:
1. Standard Energia framework
2. Yes, the same sketch runs correctly when the LaunchPad is powered by USB
3. I believe that I am providing plenty of power since I'm actually using the 3.3v output from another board that is connected to a 5VDC, 2.8A power supply. Note that I've tested with other simple sketches and they work just fine.

rei-vilo, 2015-10-22:
WiFi requires a lot of power. The 3.3V regulator of the other board may not provide enough power. 

Do you have the exact reference of the 3.3V regulator?

tom-giesberg, 2015-10-22:
I think that you may have hit on something..

The other board is a Cottonwood UHF RFID reader that is based on a Austrian Micro Systems AS3992 chip for which the datasheet states "To support external MCU and other circuitry a 3.3V regulated supply
and clock outputs are available. The regulated supply has 20mA current capability."

So I checked http://www.ti.com/lit/an/swra462/swra462.pdf and section 4.8 Current Consumption appears to state that when transmitting over WiFi, the 3200 uses 150+ mA. Am I reading that correctly? 

Do you know if there is a way, other than using the external USB, to supply 5V to the board?

rei-vilo, 2015-10-22:
Table 4.8 at chapter 8.8 rather says up to 300 mA —166~278 mA.

Have a look at the schematics [CC3200 LaunchPad Module Design Files](http://www.ti.com/lit/zip/swrc294) and especially to the `VCC_MCU_5V` line.

Again, please post this very question to the 43oh and E2E forums, as its answers would be beneficial for the whole community.

closed 2015-10-22 by rei-vilo