energia/Energia #751: LCD_SharpBoosterPack_SPI Library Constructor Parameters Question

rei-vilo, 2015-10-27:
Default constructor sets the following values for `CS`, `VCC` and `DISP`...

```
LCD_SharpBoosterPack_SPI::LCD_SharpBoosterPack_SPI() {
    LCD_SharpBoosterPack_SPI(
                 P_CS,    // Chip Select
                 P_VCC,   // Vcc display
                 P_DISP,  // DISP
                 true     // Auto VCOM
    );
}
```

... while the detailed constructor expects the parameters `CS`, `DISP` and `VCC` **in a different order**.

```
LCD_SharpBoosterPack_SPI::LCD_SharpBoosterPack_SPI(uint8_t pinChipSelect, uint8_t pinDISP, uint8_t pinVCC) {
    LCD_SharpBoosterPack_SPI(pinChipSelect, pinDISP, pinVCC, true);
}
```

Sure, both `DISP` and `VCC` are set to `HIGH` so the order has no impact on the Launchpads. 

However, for the Watch DevPack on the CC2650 SensorTag, as the implementation is different, the question may arise.

rei-vilo, 2015-10-27:
Confirmed as bug.

StefanSch, 2015-10-28:
Thanks for catching this
just made a pull request #752 with the fix.

rei-vilo, 2015-10-30:
You're welcome!

Grazfather, 2015-11-18:
Looks like it's fixed in master: c64abae3cd2600f73590931ba5187ab7c12a0b5c

StefanSch, 2015-12-02:
agreed -> closed

closed 2015-12-02 by StefanSch