energia/Energia #754: Energia 17 — Upload Fails to C2000

rei-vilo, 2015-10-30:
On Mac OS X 10.11, when trying to upload the *blinky* example to the C2000 F27028 or F27069 with **Energia 17** and the new **DSLite**, upload fails.

Additionally, there is no `sketch_oct30a.cpp.elf` file but a `sketch_oct30a.cpp.out` file instead.

Please find the error message attached.

>
/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28027.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build4158823688296389199.tmp/sketch_oct30a.cpp.elf 
Configuring Debugger (may take a few minutes on first launch)...
	Parsing connections/TIXDS100v2_Connection.xml
Failed: Cannot read System Setup data from XML file /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28027.ccxml
Error parsing file:
Fatal Error at (0, 0): An exception occurred! Type:RuntimeException, Message:The primary document entity could not be opened. Id=/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/connections/TIXDS100v2_Connection.xml while parsing file: /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/connections/TIXDS100v2_Connection.xml
>

treygerman, 2015-11-03:
Fixed in commit 1b41f145a4951f72daffc30c9d3860075ad5b609.  Please double check and close if you find the issue is resolved.

rei-vilo, 2015-11-03:
I performed an `ant clean` and then an `ant dist`.

Upload fails for either of the ports `/dev/cu.usbserial-TIY4S830B` or `/dev/cu.usbserial-TIY4S830A`.

>
Configuring Debugger (may take a few minutes on first launch)...
	Parsing connections/TIXDS100v2_Connection.xml
Failed: Cannot read System Setup data from XML file /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28069M.ccxml
Error parsing file:
Fatal Error at (0, 0): An exception occurred! Type:RuntimeException, Message:The primary document entity could not be opened. Id=/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/connections/TIXDS100v2_Connection.xml while parsing file: /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/connections/TIXDS100v2_Connection.xml
>

* `/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28069M.ccxml` does exist and is not empty, but 

* `/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/connections/TIXDS100v2_Connection.xml` does not exist.

treygerman, 2015-11-03:
Hrmm...I'm not seeing the same problem.  I notice you are doing "ant dist" but looking in the work directory.  Could this be the problem?

The connection file is there, but it is located in DSLite/common/targetdb/connections.

rei-vilo, 2015-11-03:
The `work` directory is where the compiled Energia is placed by `ant`.

rei-vilo, 2015-11-03:
For the folder, are you referring to the code source or the executable?

There is no `TIXDS100v2_Connection.xml` on the code source, but there is a `TIXDS100v3_Dot7_Connection.xml`.

treygerman, 2015-11-03:
Ahh gotcha, I was thinking "ant dist" put things in the dist directory.  The ccxml file points to "connections/TIXDS100v2_Connection.xml" but doesn't have the common/targetDB prefix.  It would appear this is some type of path issue.

rei-vilo, 2015-11-03:
The only `v2` things are `tixds100v2cortexM.xml`, `tixds100v2cs_dap.xml` and `tixds100v2icepick_c.xml`.

treygerman, 2015-11-03:
Ok...try deleting the tools-common tarball in the work/macosx directory.  When it pulls a new one you should have the file.

rei-vilo, 2015-11-03:
Wasn't the `ant clean` enough for ditching all previous zips?

treygerman, 2015-11-03:
I just checked and this tar.bz2 isn't deleted by clean

rei-vilo, 2015-11-03:
Ok, with the newly built Energia and calling
>
/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28069M.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build545040113542922883.tmp/Blink.cpp.elf 
>

* On `/dev/cu.usbserial-TIY4S830A ` port, stuck.

* On `/dev/cu.usbserial-TIY4S830B` port, error with 

>
	Mapping registers: C28xx - C
Configuring Debugger (may take a few minutes on first launch)...
	Initializing Register Database...
Failed: Operation was aborted
	Initializing: C28xx
	Executing Startup Scripts: C28xx
Connecting...
fatal: C28xx: Error connecting to the target: (Error -151 @ 0x0) One of the FTDI driver functions used during the connect returned bad status or an error. The cause may be one or more of: invalid XDS100 serial number, blank XDS100 EEPROM, missing FTDI drivers, faulty USB cable. Use the xds100serial command-line utility in the 'common/uscif' folder to verify the XDS100 can be located. (Emulation package 6.0.14.5) 
>

Two questions:
* Does the firmware of the programmer need to be updated for DSLite? 
* Any special configuration of the 3 DIP switches?

treygerman, 2015-11-03:
The port isn't used any more.  Should work with both settings.  The error you're getting it because the JTAG on the device is being held in reset.  Put the 3rd switch up and it should work.

rei-vilo, 2015-11-03:
On the F28069, DIP switches 1-2-3 = off-off-on?

treygerman, 2015-11-03:
doesn't matter at long as #3 is on

rei-vilo, 2015-11-03:
Testing on the F28027

>
Configuring Debugger (may take a few minutes on first launch)...
	Initializing Register Database...
	Initializing: C28xx
	Executing Startup Scripts: C28xx
Connecting...
fatal: C28xx: Error connecting to the target: (Error -151 @ 0x0) One of the FTDI driver functions used during the connect returned bad status or an error. The cause may be one or more of: invalid XDS100 serial number, blank XDS100 EEPROM, missing FTDI drivers, faulty USB cable. Use the xds100serial command-line utility in the 'common/uscif' folder to verify the XDS100 can be located. (Emulation package 6.0.14.5) 
Failed: Operation was aborted
>

I've tried on the two different USB ports of my MacBook and with 2 different USB cables. No luck!

Is a specific FTDI driver required?

treygerman, 2015-11-03:
Same one used on the CC3200 should work.  Are the ports showing up in the serial drop down?

rei-vilo, 2015-11-03:
Yes, they are. Upload to the CC3200 works fine.

rei-vilo, 2015-11-03:
...and upload works find with the previous solution included in **Energia 16**.

```
bash -c /Applications/IDE/Energia-16.app/Contents/Resources/Java/hardware/c2000/serial_loader2000/macos/serial_loader2000 -f /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build2930063305802925329.tmp/sketch_nov03a.cpp.txt -k /Applications/IDE/Energia-16.app/Contents/Resources/Java/hardware/c2000/F28069_flash_kernel/Debug/2806_flash_kernel.txt -b 9600 -p /dev/cu.usbserial-TIY4S830B
```

treygerman, 2015-11-03:
Hrmm very strange...and you're sure all the switches are up?

rei-vilo, 2015-11-03:
With now all switches `ON`, same result for both boards:

>
Configuring Debugger (may take a few minutes on first launch)...
Failed: Operation was aborted
	Initializing Register Database...
	Initializing: C28xx
	Executing Startup Scripts: C28xx
Connecting...
fatal: C28xx: Error connecting to the target: (Error -151 @ 0x0) One of the FTDI driver functions used during the connect returned bad status or an error. The cause may be one or more of: invalid XDS100 serial number, blank XDS100 EEPROM, missing FTDI drivers, faulty USB cable. Use the xds100serial command-line utility in the 'common/uscif' folder to verify the XDS100 can be located. (Emulation package 6.0.14.5) 
>

Have you succeeded on your side?

treygerman, 2015-11-03:
Yep, all three boards work with my Mac

rei-vilo, 2015-11-03:
Any special FTDI driver?

treygerman, 2015-11-03:
Nope :(

rei-vilo, 2015-11-03:
None of the LEDs of the programmer —blue are red on F28069; 2 red on the F28027— blinks with Energia 17, as they do with Energia 16.

treygerman, 2015-11-03:
LEDs are for serial communications only.  The new programming implementation is done via JTAG so there shouldn't be any LEDs blinking.  Still not sure why you are getting the error.

rei-vilo, 2015-11-03:
@robertinant How successful are you with the C2000s?

rei-vilo, 2015-11-04:
@treygerman

I tried again today with a fresh new binary of Energia 17 and a fresh install of the `EnergiaFTDIDrivers2.2.18.pkg` driver.

Uploading to the C2000 LaunchPads still fails on my MacBook, whatever the USB port, whatever the C2000 LaunchPad board and whatever the USB cable.

robertinant, 2015-11-06:
I have figured out what the issue is. The FTDI driver claims the JTAG interface but DSLite uses the open source libftdi driver for programming. Hence DSLite tries to open the interface but fails since the FTDI driver already claimed the interface. The solution is to NOT have the FTDI driver claim the interface by removing the JTAG interface from Info.plist in the driver. Please try the following:

```
sudo rm -rf /Library/Extensions/FTDIUSBSerialDriver.kext
```
or
```
sudo rm -rf /System/Library/Extensions/FTDIUSBSerialDriver.kext
```
Download the modified FTDI driver from: energia.nu/files/EnergiaFTDIDrivers2.3.zip
unzip and run the installer.

rei-vilo, 2015-11-06:
Congratulations for the solution!

Please package this nicely to avoid being flooded with unnecessary questions.

rei-vilo, 2015-11-06:
No luck. With the new driver, only one serial port is displayed.

Unfortunately, upload doesn't work.

Same old error message

* With F28069

>
$ /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28027.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build6911042634739637756.tmp/sketch_nov06a.cpp.elf 
Configuring Debugger (may take a few minutes on first launch)...
	Initializing Register Database...
	Initializing: C28xx
	Executing Startup Scripts: C28xx
Connecting...
fatal: C28xx: Error connecting to the target: (Error -157 @ 0x0) One of the FTDI driver functions used during start-up returned bad status or an error. (Emulation package 6.0.14.5) 
Failed: Operation was aborted
>

* With F28027

>
$
/Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28069M.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build6911042634739637756.tmp/sketch_nov06a.cpp.elf 
Configuring Debugger (may take a few minutes on first launch)...
Failed: Operation was aborted
	Initializing Register Database...
	Initializing: C28xx
	Executing Startup Scripts: C28xx
Connecting...
fatal: C28xx: Error connecting to the target: (Error -157 @ 0x0) One of the FTDI driver functions used during start-up returned bad status or an error. (Emulation package 6.0.14.5) 
>

rei-vilo, 2015-11-06:
There's another problem when compiling against the C2000 F28027. 

The required Blink.cpp.elf file isn't generated, only Blink.cpp.out and Blink.cpp.txt.

`txt` and `out` are generated...

>
/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/c2000/bin/hex2000 -boot -sci8 -a /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.out -o /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.txt 
>

... but upload requires `elf`...
>
/Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28027.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.elf 
>

...resulting in an error.

>
error: C28xx: GEL: Unable to open file: /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.elf
>

rei-vilo, 2015-11-09:
`ant clean` and `ant dist` did the trick. 

Compilation and upload are now successful for the F28027 and F28069.

Thank you!

closed 2015-11-09 by rei-vilo