energia/Energia #757: Latest OSX, Energia, and MSP432 Launchpad not working out of the box

techducky, 2015-11-06:
Energia 0101E0016 
OSX 10.11.1
MSP-EXP432P401R

Under Tools -> Board, there is no option for this board. 

When trying to upload blink to the board using the default "MSP 432 EMT": 

Connecting...
	Info: CORTEX_M4_0: GEL Output: Memory Map Initialization Complete
	Info: CORTEX_M4_0: GEL Output: Halting Watchdog Timer
Flashing program...
	Loading Program: /var/folders/h8/zjg3d5rs303dydp9r75jw4_00000gp/T/build7653244409226800129.tmp/Blink.cpp.elf:
		0% Preparing ... 
		0% .text: 0 of 25976 at 0x0
	Error: CORTEX_M4_0: Flash Programmer: Timeout error. Device init failed!
	Error: CORTEX_M4_0: File Loader: Memory write failed: Unknown error
		0% Finished
	Error: CORTEX_M4_0: GEL: File: /var/folders/h8/zjg3d5rs303dydp9r75jw4_00000gp/T/build7653244409226800129.tmp/Blink.cpp.elf: Load failed.
Failed: Failed to evaluate GEL_LoadProgramOnly( "/var/folders/h8/zjg3d5rs303dydp9r75jw4_00000gp/T/build7653244409226800129.tmp/Blink.cpp.elf" ): File: /var/folders/h8/zjg3d5rs303dydp9r75jw4_00000gp/T/build7653244409226800129.tmp/Blink.cpp.elf: Load failed.

[image]

rei-vilo, 2015-11-06:
I confirm the issue with Energia 16 on Mac OS X 10.11.

>
Initializing debug probe...
	Configuring Debugger (may take a few minutes on first launch)...:
		Initializing Register Database...
		Initializing: CS_DAP_0
		Executing Startup Scripts: CS_DAP_0
		Initializing: CORTEX_M4_0
		Executing Startup Scripts: CORTEX_M4_0
Connecting...
	Fatal: CS_DAP_0: Error connecting to the target: (Error -263 @ 0x0) Incompatible XDS110 firmware detected.  The firmware version of the connected XDS110 debug probe does not  match the expected version. Please update the firmware using the xdsdfu  utility found in  the .../ccs_base/common/uscif/xds110 directory of  your installation. View the ReadMe.txt file there for instructions. (Emulation package 5.1.630.0) 
Failed: Failed to evaluate GEL_Connect(): Connect failed
>

However, uploading works fine with Energia 17 on Mac OS X 10.11.

rei-vilo, 2015-11-06:
> Under Tools -> Board, there is no option for this board.
> When trying to upload blink to the board using the default "MSP 432 EMT":

`LaunchPad w/ MSP432 EMT (48 MHz)` is the valid option for the **MSP-EXP432P401R**.

rei-vilo, 2015-11-20:
The issue is related to the new USB management in Mac OS X 10.11.1.

rei-vilo, 2015-12-12:
[Release 17](http://energia.nu/download/) should have solved this issue.

closed 2015-12-12 by rei-vilo