energia/Energia #758: MSP430 PWM pin maps issues

StefanSch, 2015-11-06:
On several pins of the MSP430 Launchpads the PWM pins are not mapped correct:
(Thanks to Mark for identifying this)
(Sorry the text below might be a little bit scattered but gives the acutal status.)

MSP-EXP430F5529LP
Results from PWMtest2.ino with discrete LED fading. Highlight has timer in pins_energia.h. Red has failure.
Correctly fade on analogWrite()?
1		21	
2	T0B0 should be pin 42 (P1.1) [Energia Bug]	22	
3		23	
4		24	
5		25	
6		26	
7		27	
8		28	
9		29	Yes 
10		30	Yes 
11		31	
12	T2A0 properly defined according to datasheet, but failed testing [Energia Bug]	32	
13		33	
14		34	
15		35	Yes
16		36	Yes
17	Yes 	37	Yes
18		38	Yes
19	Yes	39	Yes
20		40	Yes

MSP-EXP430FR4133
Results from PWMtest2.ino with discrete LED fading. Highlight has timer in pins_energia.h. Red has failure.
Correctly fade on analogWrite()?
1	
2	
3	
4	
5	
6	
7	
8	
9	
10	T1A2 properly defined according to datasheet, but failed testing [Energia Bug]
11	
12	
13	
14	
15	
16	
17	
18	Yes
19	Yes
20	

MSP-EXP430FR5969
Results from PWMtest2.ino with discrete LED fading. Highlight has timer in pins_energia.h. Red has failure.
Correctly fade on analogWrite()?
1	
2	
3	Yes
4	T0B0 properly defined according to datasheet, but failed testing [Energia Bug]
5	
6	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
7	Yes
8	Yes
9	Yes
10	Yes
11	Yes
12	Yes
13	Yes
14	Yes
15	Yes
16	
17	
18	¬¬¬
19	Yes
20	T0B0 mislabeled, should be NOT_ON_TIMER [Energia Bug]

MSP-EXP430FR6989
Results from PWMtest2.ino with discrete LED fading. Highlight has timer in pins_energia.h. Red has failure.
Correctly fade on analogWrite()?
1		21	
2		22	
3		23	
4		24	
5		25	
6		26	
7	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]	27	
8	Strange behavior when fading? T0B6 properly defined according to datasheet, but failed testing [Energia Bug]	28	
9		29	
10		30	
11	Strange behavior when fading? T1A2 properly defined according to datasheet, but failed testing [Energia Bug]	31	
12	Yes	32	
13	Yes	33	
14	Strange behavior when fading? T0A2 properly defined according to datasheet, but failed testing [Energia Bug]	34	Yes
15	Strange behavior when fading? T0A1 properly defined according to datasheet, but failed testing [Energia Bug]	35	Strange behavior when fading? T0B4 properly defined according to datasheet, but failed testing [Energia Bug]
16		36	Strange behavior when fading? T0B3 properly defined according to datasheet, but failed testing[Energia Bug]
17		37	Strange behavior when fading? T0B2 properly defined according to datasheet, but failed testing [Energia Bug]
18	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]	38	Strange behavior when fading? T1A1 properly defined according to datasheet, but failed testing [Energia Bug]
19	T0B5 properly defined according to datasheet, but failed testing [Energia Bug]	39	Yes
20		40	Yes

MSP-EXP430G2
Results from PWMtest2.ino with discrete LED fading. Highlight has timer in pins_energia.h. Red has error.
Correctly fade on analogWrite()?
1	
2	
3	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]
4	Yes
5	
6	
7	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]
8	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
9	Strange behavior in this test– always on? T1A1 properly defined according to datasheet, and fades correctly by itself [Energia Bug]
10	Yes
11	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
12	Yes
13	Yes
14	Yes
15	
16	
17	
18	
19	Yes
20

rei-vilo, 2015-11-06:
I've already submitted the revised pins maps.

StefanSch, 2015-11-06:
just pushed the fixed code in pull request 759
Thanks Rei for updating the pin maps

StefanSch, 2015-11-06:
Just for completeness - the update comments:

MSP-EXP430F5529LP OK
2	T0B0 not there – typo with 5.6 <-> 6.5	
12	T2A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T2A0 can no generate PWM - CCR0 used as period register	

MSP-EXP430FR4133
10	T1A2 properly defined according to datasheet, but failed testing [Energia Bug]
        Fix already in previous commit

MSP-EXP430FR5969
4	T0B0 properly defined according to datasheet, but failed testing [Energia Bug]
        T0B0 can no generate PWM - CCR0 used as period register
6	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T0B0 can no generate PWM - CCR0 used as period register
20	T0B0 mislabeled, should be NOT_ON_TIMER [Energia Bug]
        fixed

MSP-EXP430FR6989OK
7	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T1A0 can no generate PWM - CCR0 used as period register	27	
8	Strange behavior when fading? T0B6 properly defined according to datasheet, but failed testing [Energia Bug]
        -> sel1 needs to be set Fixed	28	
11	Strange behavior when fading? T1A2 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed
14	Strange behavior when fading? T0A2 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed
15	Strange behavior when fading? T0A1 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed	
35	Strange behavior when fading? T0B4 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed
36	Strange behavior when fading? T0B3 properly defined according to datasheet, but failed testing[Energia Bug]
        Fixed
37	Strange behavior when fading? T0B2 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed
18	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T1A0 can no generate PWM - CCR0 used as period register	
38	Strange behavior when fading? T1A1 properly defined according to datasheet, but failed testing [Energia Bug]
        Fixed
19	T0B5 properly defined according to datasheet, but failed testing [Energia Bug]
        fixed

MSP-EXP430G2 OK
3	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T0A0 can no generate PWM
7	T0A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T0A0 can no generate PWM
8	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T1A0 can no generate PWM
9	Strange behavior in this test– always on? T1A1 properly defined according to datasheet, and fades correctly by itself [Energia Bug]
11	T1A0 properly defined according to datasheet, but failed testing [Energia Bug]
        T1A0 can no generate PWM

rei-vilo, 2015-11-10:
Other notes of interest:

* Some pins are mutually exclusive — see the [analogWrite() reference page](http://energia.nu/reference/analogwrite/).
* The timer CCR0 is already used as period register, so the PWM outputs based on CCR0 are not available.

StefanSch, 2016-01-26:
fixes merged with #759 
  fixed mapping of timer  PWMs to pins
    need to extend enums for 2 PXSEL registers.
    fixed several wrong mappings

closed 2016-01-26 by StefanSch