energia/Energia #761: Energia 17 — Wrong Executable File Compiled Against C2000

rei-vilo, 2015-11-06:
There's a problem when compiling against the C2000 F28027 and F28069. 

The required `Blink.cpp.elf` file isn't generated, only `Blink.cpp.out` and `Blink.cpp.txt`.

So upload fails. 

Chic is the correct executable to upload, `txt` or `out`?

`txt` and `out` are generated...

>
/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/c2000/bin/hex2000 -boot -sci8 -a /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.out -o /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.txt 
>

... but upload requires `elf`...
>
/Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28027.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.elf 
>

...resulting in an error.

>
error: C28xx: GEL: Unable to open file: /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build1363210983760621696.tmp/Blink.cpp.elf
>

treygerman, 2015-11-09:
I just did a git hard reset, ant clean, and ant build and I am not seeing this issue.  The elf file is correctly generated.  Check line 418 of compiler.java.  The extension of the output file should be changed to .elf.

rei-vilo, 2015-11-09:
`ant clean` and `ant dist` did the trick. 

Compilation and upload are now successful for the F28027 and F28069.

Thank you!

closed 2015-11-09 by rei-vilo