energia/Energia #762: C2000 on Mac OS X — Source Folders Contaminated with .pp Files

rei-vilo, 2015-11-06:
Another problem with the C2000 is that the compilation creates `.pp` files everywhere in the very source folders.

Those `.pp` files seems to be dependencies files. 

As a consequence, the protection mechanism built in Mac OS X monitors that `Energia.app` has changed, and flags it as corrupted. 

Other OSes may behave in a similar way.

This is not very clean. Those `.pp` files should be in the `build` directory instead, at 

>
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/
>

along with the object created

>	/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o
>

--------

Below, the content of `/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/source/F2802x_Adc.pp`

>
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/source/F2802x_Adc.c
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/F2802x_Device.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/assert.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/linkage.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/stdarg.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/stdbool.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/yvals.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/stdarg.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/_lock.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/stddef.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include/stdint.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_Adc.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_BootVars.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_DevEmu.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_Comp.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_CpuTimers.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_ECap.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_EPwm.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_Gpio.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_I2c.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_NmiIntrupt.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_PieCtrl.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_PieVect.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_Spi.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_Sci.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_SysCtrl.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_headers/include/F2802x_XIntrupt.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/include/F2802x_Examples.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/include/F2802x_GlobalPrototypes.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/include/F2802x_EPwm_defines.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/include/F2802x_I2c_defines.h
/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3793181698758132492.tmp/F2802x_Adc.c.o: /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/f2802x_common/include/F2802x_DefaultISR.h
>

rei-vilo, 2015-11-06:
```
$ cd /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000
$ find . -name \*.pp
```

gives 

>
./cores/c2000/f2802x_common/source/F2802x_Adc.pp
./cores/c2000/f2802x_common/source/F2802x_Comp.pp
./cores/c2000/f2802x_common/source/F2802x_CpuTimers.pp
./cores/c2000/f2802x_common/source/F2802x_DefaultIsr.pp
./cores/c2000/f2802x_common/source/F2802x_ECap.pp
./cores/c2000/f2802x_common/source/F2802x_EPwm.pp
./cores/c2000/f2802x_common/source/F2802x_Gpio.pp
./cores/c2000/f2802x_common/source/F2802x_I2C.pp
./cores/c2000/f2802x_common/source/F2802x_OscComp.pp
./cores/c2000/f2802x_common/source/F2802x_PieCtrl.pp
./cores/c2000/f2802x_common/source/F2802x_PieVect.pp
./cores/c2000/f2802x_common/source/F2802x_Sci.pp
./cores/c2000/f2802x_common/source/F2802x_Spi.pp
./cores/c2000/f2802x_common/source/F2802x_SWPrioritizedDefaultIsr.pp
./cores/c2000/f2802x_common/source/F2802x_SWPrioritizedPieVect.pp
./cores/c2000/f2802x_common/source/F2802x_SysCtrl.pp
./cores/c2000/f2802x_common/source/F2802x_TempSensorConv.pp
./cores/c2000/f2802x_headers/source/F2802x_GlobalVariableDefs.pp
./cores/c2000/F2806x_common/source/F2806x_Adc.pp
./cores/c2000/F2806x_common/source/F2806x_Comp.pp
./cores/c2000/F2806x_common/source/F2806x_CpuTimers.pp
./cores/c2000/F2806x_common/source/F2806x_DefaultIsr.pp
./cores/c2000/F2806x_common/source/F2806x_Dma.pp
./cores/c2000/F2806x_common/source/F2806x_ECan.pp
./cores/c2000/F2806x_common/source/F2806x_ECap.pp
./cores/c2000/F2806x_common/source/F2806x_EPwm.pp
./cores/c2000/F2806x_common/source/F2806x_EQep.pp
./cores/c2000/F2806x_common/source/F2806x_Gpio.pp
./cores/c2000/F2806x_common/source/F2806x_HRCap.pp
./cores/c2000/F2806x_common/source/F2806x_I2C.pp
./cores/c2000/F2806x_common/source/F2806x_Mcbsp.pp
./cores/c2000/F2806x_common/source/F2806x_OscComp.pp
./cores/c2000/F2806x_common/source/F2806x_PieCtrl.pp
./cores/c2000/F2806x_common/source/F2806x_PieVect.pp
./cores/c2000/F2806x_common/source/F2806x_Sci.pp
./cores/c2000/F2806x_common/source/F2806x_Spi.pp
./cores/c2000/F2806x_common/source/F2806x_SWPrioritizedDefaultIsr.pp
./cores/c2000/F2806x_common/source/F2806x_SWPrioritizedPieVect.pp
./cores/c2000/F2806x_common/source/F2806x_SysCtrl.pp
./cores/c2000/F2806x_common/source/F2806x_TempSensorConv.pp
./cores/c2000/F2806x_headers/source/F2806x_GlobalVariableDefs.pp
./cores/c2000/HardwareSerial.pp
./cores/c2000/main.pp
./cores/c2000/new.pp
./cores/c2000/Print.pp
./cores/c2000/random.pp
./cores/c2000/Stream.pp
./cores/c2000/TimerAPI.pp
./cores/c2000/TimerSerial.pp
./cores/c2000/Tone.pp
./cores/c2000/WInterrupts.pp
./cores/c2000/wiring.pp
./cores/c2000/wiring_analog.pp
./cores/c2000/wiring_digital.pp
./cores/c2000/wiring_pulse.pp
./cores/c2000/wiring_shift.pp
./cores/c2000/WMath.pp
./cores/c2000/WString.pp
>

treygerman, 2015-11-09:
Should be fixed in the latest commit

rei-vilo, 2015-11-09:
`ant clean` and `ant dist` did the trick. 

Compilation and upload are now successful for the F28027 and F28069.

Thank you!

closed 2015-11-09 by rei-vilo