energia/Energia #763: Issue uploading with MSP430, libmsp430 from command line

dazhbog, 2015-11-09:
Hi, Ive been getting this annoying error with libmsp430.

I have energia installed, i can compile and upload code fine. Using platformio OR energia's command from the command line results in the following error.

```
mspdebug tilib --force-reset "prog .pioenvs/lpmsp430f5529/firmware.hex"
MSPDebug version 0.20 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2012 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tilib: can't find libmsp430.so: dlopen(libmsp430.so, 1): image not found
scons: *** [upload] Error 255

```

MacOS 10.10.1

Any ideas?
Thanks!

robertinant, 2015-11-09:
What OS are you on? Make sure that you execute mspdebug from the actual directory that mspdebug resides.

robertinant, 2015-11-09:
```
cd Energia.app/Contents/Resources/Java/hardware/tools/msp430/mspdebug
```
and then:
```
./mspdebug tilib --force-reset "prog .pioenvs/lpmsp430f5529/firmware.hex"
```
and you should be fine.

dazhbog, 2015-11-09:
ok that worked, great, any idea/way to overcome this limitation?

robertinant, 2015-11-09:
export DYLD_LIBRARY_PATH=<path to Energia app>Energia.app/Contents/Resources/Java/hardware/tools/msp430/mspdebug

dazhbog, 2015-11-09:
Awesome! 
```
export DYLD_LIBRARY_PATH=/Applications/Energia.app/Contents/Resources/Java/hardware/tools/msp430/mspdebug
```

This worked. I also had a different version of the mspdebug and a libmsp430.so. I used the energia ones and it is now working. 

Thank you Robert.

closed 2015-11-10 by rei-vilo