energia/Energia #764: C2000 on Mac OS X 10.11.1 — Upload Stability Issues

rei-vilo, 2015-11-10:
Uploading to a C2000 from Mac OS X 10.11.1 isn't 100% reliable.

* Lost serial port after upload. `ls /dev/cu.*` lists nothing.
* `Segmentation` fault error on the F28027
* Error message on the F28377

>
/Applications/Energia.app/Contents/Resources/Java/tools/common/DSLite/DebugServer/bin/DSLite load /Applications/IDE/Energia.app/Contents/Resources/Java/tools/common/DSLite/LAUNCHXL-F28377S.ccxml /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build3116528125082285979.tmp/Blink.cpp.elf 
Configuring Debugger (may take a few minutes on first launch)...
	Initializing Register Database...
Failed: Operation was aborted
	Initializing: IcePick_C_0
	Executing Startup Scripts: IcePick_C_0
	Initializing: C28xx_CPU1
	Executing Startup Scripts: C28xx_CPU1
Connecting...
fatal: IcePick_C_0: Error connecting to the target: (Error -151 @ 0x0) One of the FTDI driver functions used during the connect returned bad status or an error. The cause may be one or more of: invalid XDS100 serial number, blank XDS100 EEPROM, missing FTDI drivers, faulty USB cable. Use the xds100serial command-line utility in the 'common/uscif' folder to verify the XDS100 can be located. (Emulation package 6.0.14.5) 
>

treygerman, 2015-11-10:
I have 10.11.1 too, but cannot recreate this issue.  I even installed the latest FTDI driver from the Energia site...still no problems on my Mac.

rei-vilo, 2015-11-10:
You're lucky! @robertinant and I have issues...

rei-vilo, 2015-11-11:
I went through another round of tests. 

* Each time the upload fails —connection lost, wrong board—, the port is no longer enumerated.

The only solution so far is a cold re-boot.

The USB management is said to be especially tricky on Mac OS X, with new bugs fixed and added at each new release.

rei-vilo, 2015-11-13:
Similar issues with CC3200. After being used, the serial-over-USB port is no longer available.

rei-vilo, 2016-01-31:
Seems to have been fixed with Mac OS X 10.11.3, although other kinds of issues with other boards have appeared.

closed 2016-01-31 by rei-vilo