energia/Energia #765: non ascii characters in source

RickKimball, 2015-11-11:
I went to edit HardwareSerial.cpp in the msp430 architecture using linux and vi. The editor barked at the non-ascii characters and truncated the file at the first non-ascii character.  There is a comment in the code that has 8 bit values.   

-               mod = divider&0xFFF0;                           // UCBRFx = INT([(N/16) <96> INT(N/16)] <D7> 16)

0x96 should be a '-' and a 0xd7 which should be a '*'

This problem has been around for a long time. It appeared in this patch:

https://github.com/energia/Energia/commit/5778939bdfce60c7fb440013f615d78ea6f6c24b#diff-1cf454b41b8d22a38557461f91553dc9R116

rei-vilo, 2015-11-11:
This is a general limitation from Processing, the IDE used by Arduino. 

Processing doesn't support extended ASCII characters.

RickKimball, 2015-11-11:
Yes, all I'm asking you to do is remove those characters.  When I went to enhanced the file with some more code. It wouldn't let me edit it without losing data.

RickKimball, 2015-11-11:
```
diff --git a/hardware/msp430/cores/msp430/HardwareSerial.cpp b/hardware/msp430/cores/msp430/HardwareSerial.cpp
index f1a6eb6..e46da54 100644
--- a/hardware/msp430/cores/msp430/HardwareSerial.cpp
+++ b/hardware/msp430/cores/msp430/HardwareSerial.cpp
@@ -121,7 +121,7 @@ void serialEventRun(void)
 // Public Methods //////////////////////////////////////////////////////////////
 #define SMCLK F_CPU //SMCLK = F_CPU for now
 
-void HardwareSerial::begin(unsigned long baud)
+void HardwareSerial::begin(unsigned long baud, uint8_t config)
 {
        unsigned int mod;
        unsigned long divider;
@@ -153,7 +153,7 @@ void HardwareSerial::begin(unsigned long baud)
                mod = ((divider&0xF)+1)&0xE;                    // UCBRSx (bit 1-3)
                divider >>=4;
        } else {
-               mod = divider&0xFFF0;                           // UCBRFx = INT([(N/16) 
<96> INT(N/16)] <D7> 16)
+               mod = divider&0xFFF0;                           // UCBRFx = INT([(N/16) - INT(N/16)] * 16)
                divider>>=8;
        }
        *(&(UCAxBR0) + uartOffset) = divider;

RickKimball, 2015-12-18:
a quick scan of the source tree reveals some more:

```
$ find . -type f -exec file {} \; | grep -i Non-ISO
./libraries/OPT3001/OPT3001.cpp: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/c2000/cores/c2000/F2806x_common/source/F2806x_Adc.c: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/c2000/cores/c2000/f2802x_common/source/F2802x_Adc.c: C source, Non-ISO extended-ASCII text
./hardware/lm4f/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/lm4f/cores/lm4f/driverlib/epi.h: C source, Non-ISO extended-ASCII text
./hardware/cc3200emt/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/cc3200/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/msp432/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/msp430/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators
./hardware/msp430/libraries/DriverLib/utility/inc/msp430f5xx_6xxgeneric.h: C source, Non-ISO extended-ASCII text
./hardware/msp430/cores/msp430/HardwareSerial.cpp: C source, Non-ISO extended-ASCII text
./emt/src/ti/runtime/wiring/cc3200/libraries/WiFi/utility/netapp.h: C source, Non-ISO extended-ASCII text, with CRLF line terminators

RickKimball, 2015-12-18:
This reveals the offending characters:

```
$ iconv -f ascii -t ascii -c ./hardware/msp430/cores/msp430/HardwareSerial.cpp >/tmp/HardwareSerial.cpp
$ diff ./hardware/msp430/cores/msp430/HardwareSerial.cpp /tmp/HardwareSerial.cpp 
156c156
< 		mod = divider&0xFFF0;                           // UCBRFx = INT([(N/16) � INT(N/16)] � 16)
---
> 		mod = divider&0xFFF0;                           // UCBRFx = INT([(N/16)  INT(N/16)]  16)
$