energia/Energia #773: ADC isn't properly powered down.

p2baron, 2015-11-22:
The ENC flag isn't cleared in the the analogread routine. ([wiring_analog.c] (https://github.com/energia/Energia/blob/master/hardware/msp430/cores/msp430/wiring_analog.c))
This prevents the ADC from being properly powered down. 
Please see [this 430oh.com] (http://forum.43oh.com/topic/9045-is-this-code-blocking-suspendlpm4/?p=69211) thread for more information.

Issue found on a MSP430G2553 using Energia 0101E0016

robertinant, 2015-11-23:
This has been fixed in Energia 17 that is soon to be released.

StefanSch, 2015-11-24:
I just checked on the latest master on github but that is special case is not fixed there now.
Maybe you have not commited/pushed the fix. So not sure if this will really go into the next Energia release.
(There have been a similar issue a while ago but not identical to that, may you have mixed that).
I can provide the fix without big effort - just let me know and i can push it.

StefanSch, 2015-12-02:
pushed fix and made pull request see  #775

StefanSch, 2015-12-02:
merged to master

closed 2015-12-02 by StefanSch