energia/Energia #774: `sleep()` proceeds garbage on Serial output.

rei-vilo, 2015-11-25:
`sleep()` proceeds garbage on Serial output on LM4F120, TM4C123 and CC3200.

I haven't tested yet on other MCUs.

>
Test ¡ÿÿzþzþzþz¢þzªþz²þzºþzÂþ)ªþªþ²þºþÂzþSH(zþHzþHzþHzþHzþHzþHzþHzþH¢zþH¢zþH¢zþH¢zþH¢zþHªzþH²zþH²zþH²zþH²zþHºzþHÂzþHÊzþHzþHzþHzþHzþHz¢þH¢z¢þHªz¢þHªz¢þH²z¢þHºz¢þHÂz¢þHÂz¢þHÂz¢þHÊz¢þHzªþHzªþHzªþHzªþHzªþHzªþH¢zªþHªzªþH²zªþ
>

Very basic sketch

```
void setup()
{
  Serial.begin(9600);
  Serial.print("Test with sleep()");
}

uint16_t a = 0;
uint16_t b = 0;

void loop()
{
  a++;
  b += (random(0, 99) % 2);

  Serial.print(b, DEC);
  Serial.print("/");
  Serial.println(a, DEC);
  sleep(10);
}
```

spirilis, 2015-11-25:
Try using Serial.flush() before your sleep().  This is known and expected behavior- sleep messes with the main CPU clock, which the UART relies upon, and all Serial print/write's are write-behind relying on asynchronous transmission after the fact that is IRQ-driven.

rei-vilo, 2015-11-25:
Thanks for the tip, it fixed the issue.

closed 2015-11-25 by rei-vilo