energia/Energia #785: boards from MSP430 family and blynk lib - compilation error on template

reaper7, 2015-12-08:
when I try to compile examples from blynk library then I get error:
```
...\libraries\blynk-library/utility/BlynkUtility.h:26:20: error: expected ')' before numeric constant
...\libraries\blynk-library/utility/BlynkUtility.h:26:20: error: expected '>' before numeric constant
...\libraries\blynk-library/utility/BlynkUtility.h:26:20: error: expected unqualified-id before ')' token
```
26 line from BlynkUtility looks:
```
template <unsigned N, typename T>
void BlynkAverageSample (T& avg, const T& input) {
    avg -= avg/N;
    const T add = input/N;
    // Fix for shorter delays
    avg += (add > 0) ? add : -1;
}
```
the same code compiled OK on other Energia platforms CC3200/Stallaris/Tiva and Arduino/ESP8266

...this is compilator error? or maybe incorrect wrote template on blynk library ?

rei-vilo, 2015-12-08:
Try and rename the variables `N` and `T` to `NN` and `TT`. 

The former names may conflict with macros.

reaper7, 2015-12-08:
LOL - it works... :D tnx!
--------------------------------
BTW
still get one more error:
```
User_Defined_Connection.cpp: In function 'size_t BlynkStreamRead(void*, size_t)':
User_Defined_Connection.cpp:42:42: error: invalid conversion from 'byte* {aka unsigned char*}' to 'char*' [-fpermissive]
C:\PROGRAMY\energia\hardware\msp430\cores\msp430/Stream.h:76:10: error:   initializing argument 1 of 'size_t Stream::readBytes(char*, size_t)' [-fpermissive]
```

rei-vilo, 2015-12-08:
* Have you renamed `T` or `N`?
* Which board are you using?
* Use transtype function to convert `byte*` to `char*`.

reaper7, 2015-12-08:
I rename T & N and error is gone:
```
template <unsigned NN, typename TT>
void BlynkAverageSample (TT& avg, const TT& input) {
    avg -= avg/NN;
    const TT add = input/NN;
    // Fix for shorter delays
    avg += (add > 0) ? add : -1;
}
```
I'm try on msp430f5529 and msp430g2955

rei-vilo, 2015-12-08:
Please report the issue and suggest the solution to Blynk! Thank you.

reaper7, 2015-12-08:
of course I report

what You mean:
"Use transtype function to convert byte* to char*."

rei-vilo, 2015-12-08:
With the following declarations
* if `function` is declared as `void function(char * value)` 
* but `myValue` as `byte* myValue;`, 
simply use
* `function((char *)myValue);`.

reaper7, 2015-12-08:
sorry but second problem is gone :)
maybe energia restart was needed.

finally: rename T to TT solves problem

closed 2015-12-08 by reaper7