energia/Energia #786: Build on linux fails due to missing tools-common-${platform}.tar.bz2 when platform is linux32 and linux64

RoXXoR, 2015-12-08:
Build fails due to missing file http://energia.nu/files/tools-common-linux64.tar.bz2
URL returns 404 Page not found and ant stops building Energia. This is true as well for http://energia.nu/files/tools-common-linux32.tar.bz2.

robertinant, 2015-12-14:
Fixed

closed 2015-12-14 by robertinant

KopeH-3JIa, 2019-03-19:
Hi @robertinant seems like this issue happens again
```

download-common-tools-linux:
     [echo] Downloading common tools
      [get] Getting: http://energia.nu/files/tools-common-linux64.tar.bz2
      [get] To: /home/vlad/Projects/solream/Energia/build/linux/tools-common.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: http://energia.nu/files/tools-common-linux64.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: http://energia.nu/files/tools-common-linux64.tar.bz2
      [get] Error opening connection java.io.FileNotFoundException: http://energia.nu/files/tools-common-linux64.tar.bz2
      [get] Can't get http://energia.nu/files/tools-common-linux64.tar.bz2 to /home/vlad/Projects/solream/Energia/build/linux/tools-common.tar.bz2

BUILD FAILED
/home/vlad/Projects/solream/Energia/build/build.xml:94: The following error occurred while executing this line:
/home/vlad/Projects/solream/Energia/build/build.xml:82: The following error occurred while executing this line:
/home/vlad/Projects/solream/Energia/build/build.xml:490: Can't get http://energia.nu/files/tools-common-linux64.tar.bz2 to /home/vlad/Projects/solream/Energia/build/linux/tools-common.tar.bz2
```
[tools-common-linux64.tar.bz](http://energia.nu/files/tools-common-linux64.tar.bz2) return  404