energia/Energia #795: Add Documentation and Examples for Low Power Mode

rei-vilo, 2015-12-12:
Add documentation and examples for low power mode functions.

* Functions: `sleepSeconds()`, `suspend()` and `wakeup()`
* Documentation: section **More on Energy Savings** is hidden below [How to measure Energy profile in Energia with TI Energy Trace](http://energia.nu/guide/tutorial_energytrace/) to be posted on a full page.
* Example: [fmilburn3/Energia_LPM](https://github.com/fmilburn3/Energia_LPM), to be included in the examples provided with Energia.

rei-vilo, 2015-12-29:
See http://embeddedcomputing.weebly.com/ultra-low-power-with-msp430.html

spirilis, 2015-12-29:
@rei-vilo Might be worth mentioning in delay vs. sleep that as a consequence of SMCLK disabled, background processes such as Serial transmit and receive will halt or get scrambled.

rei-vilo, 2015-12-29:
@spirilis Thanks! Added!