energia/Energia #797: msp430: i2c slave does not handle repeated start

itiserik, 2015-12-14:
When on arduino the Wire library is used as an i2c slave. And a master (e.g. i2cget on linux) is used to query the slave, both callbacks are called: onReceive at the repeated start and onRequest when the repeated start frame is ack'ed. See also arduino#66.

On the MSP430 the i2c interrupts are slightly different. The hardware does not detect the repeated start and the library does not call the onReceive callback.

rei-vilo, 2015-12-14:
Which I²C port are you using?

* on pins 15 + 16 —hardware master + slave—, or 
* on pins 9 + 10 —software emulated master only.

itiserik, 2015-12-15:
I use the hardware I2C port on the msp430G2553 as slave. The launchpad is connected using breadboard wires to an at-91 based cpu running debian. On that host I try to query the slave using i2cget \<bus> \<chip address> \<data address> b

Its about the data address here, I would like to receive this on the msp and interpret it as a command or a register offset. i2cget does this by calling i2c_smbus_read_byte_data() read here about this function https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/Documentation/i2c/smbus-protocol

On arduino the stop condition flag is set on stop OR repeated start arduino then calls onReceive (with the data address sent by i2cget). When the slave becomes addressed again the ack flag is set and onRequest is called.

By calling i2cget in this way I can query the device for only one byte at an specified offset instead of streaming all bytes to the host and ignoring the not needed bytes.

rei-vilo, 2015-12-15:
There are differences between I²C and SMBus.

I²C issue might be related to #789.

StefanSch, 2015-12-16:
The fix in #789 only applies for eUSCI devices. The G2553 has USCI and is not affected by this issue.

itiserik, 2015-12-16:
After updating my Energia to 0101E0017 and applying #789 again. I discovered that the i2c slave on my Launchpad did not work. After inspection I came up with #799.

Differences between I2C and SMBus are neglectable and have very sure nothing to do with this.