energia/Energia #800: MSP430: Provide Documentation on New I²C Implementation

rei-vilo, 2015-12-18:
Please provide documentation on the new I²C implementation.

To ensure compatibility with the BoosterPack standard, default I²C port is now on pins `9` and `10`. 

* For MCUs with no hardware I²C port on pins `9` and `10`, I²C port on pins `9` and `10` is software emulated and runs only as master.

* For MCUs with hardware I²C port on pins `15` and `16`, I²C port on pins `15` and `16` runs as either master and slave.  

See [Using HW (not SW) I²C](http://forum.43oh.com/topic/9176-using-hw-not-sw-i2c)

rei-vilo, 2015-12-18:
See `TwoWire(uint8_t module);` at https://github.com/energia/Energia/blob/master/hardware/msp430/cores/msp430/Wire.h

rei-vilo, 2015-12-18:
Maybe @StefanSch can help. 

I don't have access to edit the pages at [energia.nu](energia.nu).

rei-vilo, 2015-12-18:
I've updated the pins maps for the impacted boards but they haven't been posted yet.

itiserik, 2015-12-19:
Ah the boosterpack standard causes this broken i2c interface (no slave).
It would have been nicer when there was a seperate SoftwareMasterTwoWire library with configurable pins. When a slave is implemented this lib can be renamed to SoftwareTwoWire, analogue to SoftwareSerial.

Advantage of this is that the existing slave wire examples will keep working because they keep using the hw I2C. And most importantly that the twi.c does not get cluttered with uneeded defines about the sw implementation.

Boosterpack examples should have a way to find out to which launchpad they are connected. And then instanciate the correct Wire object (i.e. TwoWire(0 or 1) or SoftwareMasterTwoWire(SCL,SDA).

rei-vilo, 2015-12-19:
The SoftwareMasterTwoWire library exists. 

See [I²C_SoftwareLibrary](https://github.com/rei-vilo/I2C_SoftwareLibrary)

rei-vilo, 2015-12-19:
@measley Please update the pins maps with release 2.x.