energia/Energia #802: crash when starting from Linux

elpaso, 2015-12-18:
Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException: No such child: 3
        at java.awt.Container.getComponent(Container.java:334)
        at javax.swing.JPopupMenu.insert(JPopupMenu.java:595)
        at javax.swing.JMenu.insert(JMenu.java:706)
        at processing.app.Editor$2.windowActivated(Editor.java:180)
        at java.awt.AWTEventMulticaster.windowActivated(AWTEventMulticaster.java:390)
        at java.awt.AWTEventMulticaster.windowActivated(AWTEventMulticaster.java:389)
        at java.awt.Window.processWindowEvent(Window.java:2070)
        at javax.swing.JFrame.processWindowEvent(JFrame.java:305)
        at java.awt.Window.processEvent(Window.java:2017)
        at java.awt.Component.dispatchEventImpl(Component.java:4891)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Window.dispatchEventImpl(Window.java:2750)
        at java.awt.Component.dispatchEvent(Component.java:4713)
        at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
        at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:995)
        at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:488)
        at java.awt.Component.dispatchEventImpl(Component.java:4762)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Window.dispatchEventImpl(Window.java:2750)
        at java.awt.Component.dispatchEvent(Component.java:4713)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.SentEvent.dispatch(SentEvent.java:70)
        at java.awt.DefaultKeyboardFocusManager$DefaultKeyboardFocusManagerSentEvent.dispatch(DefaultKeyboardFocusManager.java:217)
        at java.awt.DefaultKeyboardFocusManager.sendMessage(DefaultKeyboardFocusManager.java:244)
        at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:366)
        at java.awt.Component.dispatchEventImpl(Component.java:4762)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Window.dispatchEventImpl(Window.java:2750)
        at java.awt.Component.dispatchEvent(Component.java:4713)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.SequencedEvent.dispatch(SequencedEvent.java:128)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

rei-vilo, 2015-12-18:
Could please specify:

* Which version of Energia? 
* Which distort of Linux?

elpaso, 2015-12-18:
Energia: latest energia-0101E0017
Linux: Ubuntu 14.04 64bit

Update: the second time I launch it, it doesn't show a stacktrace anymore but it's deadly slow: completely unusable, I had to switch back to energia-0101E0013 (that works fine).

spirilis, 2015-12-18:
random question, what size display/monitor/resolution are you running?

elpaso, 2015-12-18:
4k monitor 282dpi  3840 x 2160

spirilis, 2015-12-18:
wow, ok, definitely not a "lack of screen height" issue like I've seen with some! (there was a bug where the Examples folder in File menu couldn't be calculated correctly based on the # of entries relative to the size of your screen, but that tends to throw a Java error)

spirilis, 2015-12-18:
Second question, are you using the built-in Java or an official Oracle Java release?  I've been using official Java 8 releases for the past 4 or 5 versions I think... I don't touch the Ubuntu-provided stuff anymore.

elpaso, 2015-12-18:
java -version
java version "1.8.0_66"
Java(TM) SE Runtime Environment (build 1.8.0_66-b17)
Java HotSpot(TM) 64-Bit Server VM (build 25.66-b17, mixed mode)

RickKimball, 2015-12-18:
FWIW: I'm running ubuntu 14.04 64 bit and it runs fine. I have the same configuration as you elpaso .. using java 1.8.0_66

RickKimball, 2015-12-18:
@elpaso  you might want to run energia like this .. to see if there is anything strange about your setup

$ cd energia install dir
$ bash -x ./energia

post your output here

RickKimball, 2015-12-18:
@elpaso another question .. which launchpad are you using? I'm using the msp430fr5969

elpaso, 2015-12-21:
```
ale@darkstar ~/energia-0101E0017: bash -x ./energia
+ export PATH=/home/ale/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/ale/bin:/opt/mspgcc/bin
+ PATH=/home/ale/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/ale/bin:/opt/mspgcc/bin
+++ readlink -f -- ./energia
++ dirname -- /home/ale/energia-0101E0017/energia
+ APPDIR=/home/ale/energia-0101E0017
+ cd /home/ale/energia-0101E0017
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar:lib/core.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar:lib/core.jar:lib/ecj.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar:lib/core.jar:lib/ecj.jar:lib/jna.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar:lib/core.jar:lib/ecj.jar:lib/jna.jar:lib/pde.jar
+ for LIB in java/lib/rt.jar java/lib/tools.jar 'lib/*.jar'
+ CLASSPATH=:java/lib/rt.jar:java/lib/tools.jar:lib/core.jar:lib/ecj.jar:lib/jna.jar:lib/pde.jar:lib/RXTXcomm.jar
+ export CLASSPATH
++ pwd
+ LD_LIBRARY_PATH=/home/ale/energia-0101E0017/lib
+ export LD_LIBRARY_PATH
+ export PATH=/home/ale/energia-0101E0017/java/bin:/home/ale/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/ale/bin:/opt/mspgcc/bin
+ PATH=/home/ale/energia-0101E0017/java/bin:/home/ale/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/ale/bin:/opt/mspgcc/bin
+ java -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel processing.app.Base
Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
```

I tested with both stellaris ans LP with 2553

Strange enough, I tested it right now and the traceback is gone.
The traceback appeared only when launching energia the first time, the slowness seems also to be gone.

robertinant, 2015-12-28:
Seems like you got things to work. Can you please confirm so that we can close this issue?

elpaso, 2015-12-29:
Hi, the problem is solved for me but I suspect that there is still something wrong: I don't know if there is something clashing with my previous energia installations or what is the  cause of the initial traceback (I'm not a Java dev, sorry).
You should try to reproduce this issue on a clean ubuntu installation, I'm not scared about tracebacks but some other users might be discouraged and just quit with a bad impression about this software.
The reason for the traceback and the initial slowness is stil a mistery.

rei-vilo, 2015-12-29:
I hope next release of Energia will embed JRE 8 as Arduino 1.6.7 does to avoid JRE 6.

See Readiness for Java SE 8 #533

robertinant, 2015-12-29:
@elpaso, I will be moving the Energia IDE to the 1.6.x Arduino IDE which, as @rei-vilo mentioned, embeds it's own JRE.

closed 2015-12-29 by robertinant