energia/Energia #803: Test LiquidCrystal Library for platforms other than MSP430

robertinant, 2015-12-18:
Right now the LiquidCrystal Library is only visible for MSP430. Testing needs to be done for the other platforms so that it can be included for those platforms as well.

fmilburn3, 2015-12-19:
I have tested it with "Hello World" on the MSP432 and the TMC123GXL successfully.  See this post:  http://forum.43oh.com/topic/9186-hello-world-using-coglcd/?p=69914

KingZuluKing, 2017-08-05:
Hey guys, does anyone know if there is a liquid Crystal library  for HD44780 displays which works with tm4c 129EX launchpad?

rei-vilo, 2017-08-05:
Please check on the 43oh forum.

KingZuluKing, 2017-08-05:
thanks, by the way, what do you think a msp430 library of the same LiquidCrystal will not work on tiva -tm4c 129Xl launchpad? if I import it into suitable folder. Because I did the same for C2000 Piccolo Launchpad, and it worked without any issue.

rei-vilo, 2017-08-05:
See https://github.com/energia/Energia/issues/803#issuecomment-166010139, if it works on TM4C123, it will work on TM4C129.

KingZuluKing, 2017-08-05:
checked right now, drops the following errors... maybe you can see something i am mising out here

--------------------------------------------------------------------------------
sketch_aug05a.cpp.o: In function `setup':
sketch_aug05a.cpp:(.text.setup+0x8): undefined reference to `LiquidCrystal::begin(unsigned char, unsigned char, unsigned char)'
sketch_aug05a.cpp.o: In function `_GLOBAL__sub_I_lcd':
sketch_aug05a.cpp:(.text.startup._GLOBAL__sub_I_lcd+0x14): undefined reference to `LiquidCrystal::LiquidCrystal(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
collect2.exe: error: ld returned 1 exit status

KingZuluKing, 2017-08-05:
Tested , works 100 !

robertinant, 2017-08-30:
This issue was moved to energia/tivac-core#25

closed 2017-08-30 by robertinant