energia/Energia #804: Can't build from source

RickKimball, 2015-12-18:
See conversation starting here:

http://forum.43oh.com/topic/9176-using-hw-not-sw-i2c/?p=69895

rei-vilo, 2015-12-18:
Please, which OS?

RickKimball, 2015-12-18:
linux 32 bit

RickKimball, 2015-12-18:
Question, which version of java is used to compile? I'm using java from oracle version 

$ java -version
java version "1.8.0_66"

RickKimball, 2015-12-18:
FWIW: building master on linux 64 seems better than the 32 bit version.. I rebuilt from HEAD and it looks like only recent changes are the difference

robertinant, 2015-12-19:
Thanks @RickKimball! Happy to hear that the 64 bit build looks better. The tag 0101E0017-emt-a61 should represent the release but it is possible that some local changes snug into the release before they were committed.

RickKimball, 2015-12-19:
@robertinant I checked out that tag and built it on linux 64 bit. It doesn't match. Can you checkout that tag in a new area, build and compare to see if that tag is real the same.  If they turn out to be can you let me know exactly what steps are performed?

robertinant, 2015-12-21:
I just checked both my 64 and 32 bit Linux machines and both of the were at f3c5a03 when I build the release. My working directory is clean so I have no explanation for the differences. This commit ID is the same as the 0101E0017-emt-a61. I will do some more digging to try and figure out what is going on.

For now, please use master for the enhancements you would like to contribute.

RickKimball, 2015-12-22:
So if I checkout master I get this:

```
BUILD FAILED
/home/kimballr/Energia/build/build.xml:81: The following error occurred while executing this line:
/home/kimballr/Energia/build/build.xml:449: The following error occurred while executing this line:
/home/kimballr/Energia/build/build.xml:177: The following error occurred while executing this line:
/home/kimballr/Energia/build/build.xml:64: gunzip is not a legal value for this attribute

Total time: 7 seconds
```

Looking at the file closer it turns out it isn't a gunzip file.
```
i386:$ file linux/closure-emt-a61.tar.gz 
linux/closure-emt-a61.tar.gz: POSIX tar archive
(reverse-i-search)`u': file linux/clos^Ce-emt-a61.tar.gz 
i386:$ gunzip linux/closure-emt-a61.tar.gz 

gzip: linux/closure-emt-a61.tar.gz: not in gzip format
```

lohmanndouglas, 2016-03-19:
If you do like RickKimball says and de file is not in GZIP format. 

You try rename the file linux/closure-emt-a61.tar.gz  to linux/closure-emt-a61.tar them create a tar.gz with gzip command. ($ gzip linux/closure-emt-a61.tar  ).

Now you can build 

I have the same problem and It woks for me.

RickKimball, 2016-04-13:
The answer is probably related to the TI build environment not matching the checked in source code.
See here:
https://github.com/energia/Energia/commit/8c25e7c5d6f20df6d4021e1497fb2ac46a35be32

closed 2016-04-13 by RickKimball