energia/Energia #807: MCU Wrong Definition for EK-TM4C123GXL LaunchPad

rei-vilo, 2015-12-19:
File `Energia.app/Contents/Resources/Java/hardware/lm4f/cores/lm4f/Energia.h` includes a wrong definition for the EK-TM4C123GXL LaunchPad.

```
#if defined(__TM4C129XNCZAD__)
#define TARGET_IS_SNOWFLAKE_RA0
#define PART_TM4C129XNCZAD
#elif defined(__TM4C1294NCPDT__)
#define TARGET_IS_SNOWFLAKE_RA0
#define PART_TM4C1294NCPDT
#elif defined(__LM4F120H5QR__) || defined(__TM4C123GH6PM__)
#define TARGET_IS_BLIZZARD_RB1
#define PART_TM4C1233H6PM 
#define PART_LM4F120H5QR
#else
#error "**** No PART defined or unsupported PART ****"
#endif
```

It should be `PART_TM4C123GH6PM` instead of `PART_TM4C1233H6PM` as per [ARM® Cortex®-M4F Based MCU TM4C123G LaunchPad™ Evaluation Kit](http://www.ti.com/tool/ek-tm4c123gxl) product page.

>
The TM4C123G LaunchPad Evaluation Kit is a low-cost evaluation platform for ARM Cortex-M4F based microcontrollers from Texas Instruments. The design of the TM4C123G LaunchPad highlights the TM4C123GH6PM microcontroller with a USB 2.0 device interface and hibernation module.
>

As a consequence, wrong pins are selected in `Energia.app/Contents/Resources/Java/hardware/lm4f/cores/lm4f/driverlib/pin_map.h`. Same issue arises for the compatible Stellaris LaunchPad based on the LM4F120H5QR MCU.

rei-vilo, 2015-12-19:
[image]

spirilis, 2015-12-19:
Fwiw, the part listed "PART_TM4C1233H6PM" correctly corresponds to the Tiva-C branded equivalent for the Stellaris LM4F120H5QR.  It's Tiva-C TM4C123G that needs its part entry changed.

rei-vilo, 2015-12-19:
I don't understand. Is the specification sheet of the TM4C123G LaunchPad wrong?

>
It should be `PART_TM4C123GH6PM` instead of `PART_TM4C1233H6PM` as per [ARM® Cortex®-M4F Based MCU TM4C123G LaunchPad™ Evaluation Kit](http://www.ti.com/tool/ek-tm4c123gxl) product page.
>>
The TM4C123G LaunchPad Evaluation Kit is a low-cost evaluation platform for ARM Cortex-M4F based microcontrollers from Texas Instruments. The design of the TM4C123G LaunchPad highlights the TM4C123GH6PM microcontroller with a USB 2.0 device interface and hibernation module.
>>

The TM4C123G doesn't feature QEI, and I've just tested QEI on both LM4F120H5- and TM4C123GH6PM-based LaunchPads successfully.

rei-vilo, 2015-12-20:
The LM4F120H5QR doesn't have QEI...

>
SysCtlPeripheralPresent
SYSCTL_PERIPH_QEI0      0
SYSCTL_PERIPH_QEI1      0
>

...while the TM4C123GH6PM does feature it...

>
SysCtlPeripheralPresent
SYSCTL_PERIPH_QEI0      1
SYSCTL_PERIPH_QEI1      1
>

...thanks to this very basic sketch based on [16.2.1.23 ROM_SysCtlPeripheralPresent from LM4F120H5QR ROM USER’S GUIDE (spmu245a)](http://www.ti.com/lit/ug/spmu245a/spmu245a.pdf).

```
#include "Energia.h"
#include "driverlib/sysctl.h"

void setup()
{
    Serial.begin(9600);
    delay(250);
    Serial.println("SysCtlPeripheralPresent");
    
    Serial.print("SYSCTL_PERIPH_QEI0\t");
    Serial.println(SysCtlPeripheralPresent(SYSCTL_PERIPH_QEI0));
    Serial.print("SYSCTL_PERIPH_QEI1\t");
    Serial.println(SysCtlPeripheralPresent(SYSCTL_PERIPH_QEI1));
}

void loop()
{
}
```

rei-vilo, 2015-12-20:
At the same time, `Energia.app/Contents/Resources/Java/hardware/lm4f/variants/stellarpad/part.h` defines for LM4F120H5QR- and TM4C123GH6PM-based LaunchPad 

```
#define __LM4F120H5QR__
#define __TM4C123GH6PM__
```

but `Energia.app/Contents/Resources/Java/hardware/lm4f/cores/lm4f/Energia.h` doesn't converted `__TM4C123GH6PM__ ` into `PART_TM4C123GH6PM`, hence the issues I'm facing with QEI for example.

rei-vilo, 2015-12-20:
The previous gives 1 QEI for the TM4C129ENCPDT *aka.* Connected LaunchPad.

>
SYSCTL_PERIPH_QEI
SYSCTL_PERIPH_QEI0      1
SYSCTL_PERIPH_QEI1      0
>

rei-vilo, 2015-12-20:
The boards flags defined by Energia 0101E0017 are

* **LM4F120** `__LM4F120H5QR__` `__TM4C123GH6PM__`
* **TM4C123** `__LM4F120H5QR__` `__TM4C123GH6PM__`
* **TM4C129** `__TM4C1294NCPDT__`

robertinant, 2017-08-30:
This issue was moved to energia/tivac-core#24

closed 2017-08-30 by robertinant