energia/Energia #808: UnsatisfiedLinkError starting energia-0101E0017-linux64 on Ubuntu 15.10 (64 bit)

tschutter, 2015-12-24:
After selecting the sketch folder:

    OpenJDK Server VM warning: You have loaded library /home/tom/energia-0101E0017/lib/librxtxSerial.so which might have disabled stack guard. The VM will try to fix the stack guard now.
    It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
    java.lang.UnsatisfiedLinkError: /home/tom/energia-0101E0017/lib/librxtxSerial.so: /home/tom/energia-0101E0017/lib/librxtxSerial.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch) thrown while loading gnu.io.RXTXCommDriver
    Exception in thread "main" java.lang.UnsatisfiedLinkError: /home/tom/energia-0101E0017/lib/librxtxSerial.so: /home/tom/energia-0101E0017/lib/librxtxSerial.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary1(ClassLoader.java:1965)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1890)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1880)
	at java.lang.Runtime.loadLibrary0(Runtime.java:849)
	at java.lang.System.loadLibrary(System.java:1088)
	at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:83)
	at processing.app.Editor.populateSerialMenu(Editor.java:1085)
	at processing.app.Editor.buildToolsMenu(Editor.java:804)
	at processing.app.Editor.buildMenuBar(Editor.java:478)
	at processing.app.Editor.<init>(Editor.java:206)
	at processing.app.Base.handleOpen(Base.java:778)
	at processing.app.Base.handleOpen(Base.java:743)
	at processing.app.Base.handleNew(Base.java:594)
	at processing.app.Base.<init>(Base.java:331)
	at processing.app.Base.main(Base.java:205)

This sounds similar to #98.

robertinant, 2015-12-28:
hmm, I though this was taken care of in the build process. The issue is that by default the 32 bit version of the librxtx library is included. The build process to create Energia is suppose to take care of including the 64-bit version. You can work around this issue by doing:

```
cd /home/tom/energia-0101E0017/lib/
mv librxtxSerial.so librxtxSerial32.so
mv librxtxSerial64.so librxtxSerial.so
```
Please let me know if that worked.

Robert

robertinant, 2015-12-28:
I am going to keep this open so that it get's fixed before the next release.

tschutter, 2015-12-31:
The workaround doesn't solve the problem because librxtxSerial.so and librxtxSerial64.so are identical.

    $ cmp librxtxSerial.so librxtxSerial64.so
    $

hemangjoshi37a, 2017-05-01:
stops running 	at processing.app.Base.main(Base.java:205)