energia/Energia #811: Energia not honoring #ifdef preprocessor directives when linking libraries

axsdenied, 2015-12-27:
When creating a list of libraries for linking, Energia seems to add all #include statements, without actually checking #ifdef c preprocessor directives. 

For example, trying to write one sketch for multiple devices:
#define _TIVA
#ifdef _TIVA
#include "A.h"
#else
#include "B.h"
#endif
will try to link both A and B libraries at the same time and possibly create conflict. 

Also mentioned here:
http://forum.43oh.com/topic/6284-how-to-select-correct-include-file/

rei-vilo, 2015-12-27:
This is a referenced bug from the Processing —*a.k.a. Arduino— IDE.

Place the critical `#ifdef` `#else` `#end` pre-processing statements on a library.

closed 2015-12-27 by rei-vilo

robertinant, 2015-12-28:
This is an Energia/Arduino pre-processor "bug"/"feature" depending how you look at it. The Energia/Arduino IDE parses the Sketch looking for #include's of libraries to collect and add them to the list of files to compile. If you #include libraries that have the same function's, defines, macro's, etc you end up with the linker error you see.
 
There is currently no way around this other then creating one single library that supports all chips you which to support and use the defines given to you by Energia to make decisions for chip specific functions.
 
Hope this helps.

axsdenied, 2016-01-06:
Not quite, this bug has been fixed in Arduino v1.6.6 but it still exists in Energia.
See
https://github.com/arduino/Arduino/issues/906
and the fix here:
https://github.com/arduino/Arduino/pull/3779

It would be great if we could get the fix as well.

rei-vilo, 2016-01-06:
Energia 0101E017 is based on Arduino 1.0.

See ticket Update the IDE to release 1.6.7 of Arduino #588.