energia/Energia #814: BMA222 Library Interferes with YELLOW and GREEN LEDs on CC3200 Launchpad

craigfoo, 2015-12-31:
When I use the BMA222 library in conjunction with using the on-board LEDs, issues start happening. It only seems to be apparent using the YELLOW and GREEN LEDs but each have a different behavior. 

RED LED, everything works fine
YELLOW LED, program gets stuck in I2C loop
GREEN LED, no accelerometer data

It only happens when I set the pinmode for the LED so something is conflicting there. Here is my test program based on the BMA222 example:

    #include <Wire.h>
    #include <BMA222.h>

    BMA222 mySensor;

    static uint8_t led = RED_LED;
    //static uint8_t led = YELLOW_LED;
    //static uint8_t led = GREEN_LED;

    void setup()
    {
      Serial.begin(115200);
  
      mySensor.begin();
      uint8_t chipID = mySensor.chipID();
      Serial.print("chipID: ");
      Serial.println(chipID);

      pinMode(led, OUTPUT);
      digitalWrite(led, LOW);
    }

    void loop()
    {

      int8_t data = mySensor.readXData();
      Serial.print("X: ");
      Serial.print(data);

      data = mySensor.readYData();
      Serial.print(" Y: ");
      Serial.print(data);

      data = mySensor.readZData();
      Serial.print(" Z: ");
      Serial.println(data);

      digitalWrite(led, HIGH);
      delay(250);
      digitalWrite(led, LOW);
      delay(250);

    }

I'll try to track it down and update here if I find anything.

rei-vilo, 2015-12-31:
Please have a look at the schematics of the CC3200 LaunchPad.

craigfoo, 2015-12-31:
I see now that the YELLOW and GREEN LEDs are on the same GPIO as the SDA and SCL lines. That's the problem, bummer. Break out the soldering iron, need to make a modification.

closed 2015-12-31 by rei-vilo

rei-vilo, 2015-12-31:
Are the 2 jumpers in place? I have no issues on mine. 

[image]