energia/Energia #815: min() and max() not declared with EMT CC3200 & MSP432

kingofl337, 2016-01-02:
When using EMT version of the CC3200 calling min() and max() results in a unknown declaration, using non-emt board works fine.

code
void setup()
{
  // put your setup code here, to run once:
  
}

void loop()
{
  // put your main code here, to run repeatedly:
  int x = min(3,5);
  int y = max(5,7);
}

sketch_jan02c.ino:14:18: error: 'min' was not declared in this scope
sketch_jan02c.ino:15:18: error: 'max' was not declared in this scope

rei-vilo, 2016-01-02:
See #815

robertinant, 2016-01-04:
Thanks for the report!
This has been fixed and will be in the next release.
For now you can work around it by adding the macros at the top of your Sketch.
```
#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))
```

closed 2016-01-04 by robertinant